tesseract-robotics/tesseract #380: Modifying kinematic limits during execution

mpowelson, 2020-09-08:
I want to do something like this.
```
if (mode == "SPECIAL_PLANNING_CASE")
  kin->setLimits(...);
```

I seems like it should be as simple as just adjusting the kinematic limits in the kdl data structure. Do you think there are other methods that will have to be called again? For instance, would Tesseract objects for example have to be reinitialized? I'm thinking the answer is no since I don't see anywhere where kinematics gets processed into another data structure, but I wanted to see if anyone else (@Levi-Armstrong) knew of anything.

Levi-Armstrong, 2020-09-08:
It only needed at the TesseractKinematics object or are you also wanting to modify it in environment ?

mpowelson, 2020-09-08:
I want to modify it everywhere needed for planning. So I suppose in the environment as well.

Levi-Armstrong, 2020-09-08:
I would create a new command and apply it that way, should be to much work. Maybe ModifyJointLimitsCommand?

mpowelson, 2020-09-09:
Ok. Depending on if I end up needing to do this, I may add that later this week. I'll assign this to myself if I start it.

mpowelson, 2020-10-14:
@Levi-Armstrong I think I am going to take a stab at this.

mpowelson, 2020-10-19:
See #415

Levi-Armstrong, 2020-11-12:
Integrated in #415.

closed 2020-11-12 by Levi-Armstrong