tesseract-robotics/tesseract #381: Time parameterization returning trajectories that are not monotonically increasing when given duplicate points

mpowelson, 2020-09-08:
It appears that ISP is returning trajectories with points that have the same time_from_start. Here is an example of results
```
points[]
  points[0]: 
    positions[]
      positions[0]: 0.0529602
      positions[1]: 2.99159
      positions[2]: 0.07
      positions[3]: 0.07
      positions[4]: 0
    velocities[]
      velocities[0]: 0
      velocities[1]: 0
      velocities[2]: 0
      velocities[3]: 0
      velocities[4]: 0
    accelerations[]
      accelerations[0]: 0
      accelerations[1]: 0
      accelerations[2]: 0
      accelerations[3]: -9.91795
      accelerations[4]: 0
    effort[]
    time_from_start: 0.000000000
  points[1]: 
    positions[]
      positions[0]: 0.0529602
      positions[1]: 2.99159
      positions[2]: 0.07
      positions[3]: 0.07
      positions[4]: 0
    velocities[]
      velocities[0]: 0
      velocities[1]: 0
      velocities[2]: 0
      velocities[3]: 3.38705e-08
      velocities[4]: 0
    accelerations[]
      accelerations[0]: 0
      accelerations[1]: 0
      accelerations[2]: 0
      accelerations[3]: 19.8455
      accelerations[4]: 0
    effort[]
    time_from_start: 0.000000003
  points[2]: 
    positions[]
      positions[0]: 0.0529602
      positions[1]: 2.99159
      positions[2]: 0.07
      positions[3]: 0.07
      positions[4]: 0
    velocities[]
      velocities[0]: 0
      velocities[1]: 0
      velocities[2]: 0
      velocities[3]: 3.85246e-08
      velocities[4]: 0
    accelerations[]
      accelerations[0]: 0
      accelerations[1]: 0
      accelerations[2]: 0
      accelerations[3]: -7.15299
      accelerations[4]: 0
    effort[]
    time_from_start: 0.000000003
```

This seems like a good thing except for the fact that velocities and accelerations are different. However, maybe we should modify it to require some small amount of time between waypoints? Ros_control doesn't like it when you pass a trajectory like this. 

Is there a generic way this is usually handled? It looks like I am probably commanding 2 duplicated points when planning as 10 points in a row have the same positions (which should correspond to the 10 points in the seed associated with the first move). I would hope that wouldn't break things, but maybe we should have some way of checking for that and at least printing a warning.

gavanderhoorn, 2020-09-08:
I would be more worried about that instanteneous acceleration of `-9.91795` rad/s^2 of joint 3 there at `time_from_start == 0.0`.

mpowelson, 2020-09-08:
Yeah. That's concerning as well. It seems that in general bad things happen if they are on top of each other. Which probably isn't too surprising

gavanderhoorn, 2020-09-08:
> Which probably isn't too surprising

Splines with pts being very close to each other is not a good idea, no.

Levi-Armstrong, 2020-12-11:
@mpowelson You good with closing this issue or moving it to the discussions?

closed 2020-12-12 by mpowelson