tesseract-robotics/tesseract #391: Odd behavior setting safety margin vector

mpowelson, 2020-09-25:
How confident are we that setting the SafetyMarginData in our trajopt planner is getting correctly propagated? Even doing this like below I am getting collision distances reported unless I actually change the default values in trajop_collision_config.h. Perhaps I am doing something obviously wrong. Below does not work, but changing the default values (which I should be changing below) does. However, that obviously doesn't let you set just a single value like I was wanting to do.

```
    auto trajopt_planner = std::make_shared<TrajOptMotionPlanner>();
    trajopt_planner->problem_generator = &DefaultTrajoptProblemGenerator;
    trajopt_planner->plan_profiles["FREESPACE"] = std::make_shared<TrajOptDefaultPlanProfile>();
    auto composite_profile = std::make_shared<TrajOptDefaultCompositeProfile>();

    auto cost_safety_margin = std::make_shared<trajopt::SafetyMarginData>(0.00, 20);
    cost_safety_margin->setPairSafetyMarginData("link_1", "link_2", 0.0, 0);
    composite_profile->special_collision_cost = cost_safety_margin;

    auto cnt_safety_margin = std::make_shared<trajopt::SafetyMarginData>(0.0, 20); 
    cnt_safety_margin->setPairSafetyMarginData("link_1", "link_2", 0.0, 10);

    composite_profile->special_collision_constraint = cnt_safety_margin;
    composite_profile->collision_constraint_config.safety_margin_buffer = 0.0;
    composite_profile->collision_cost_config.safety_margin_buffer = 0.0;
    composite_profile->collision_constraint_config.safety_margin = 0.0;
    composite_profile->collision_cost_config.buffer_margin = 0.0;

    composite_profile->collision_cost_config.type = trajopt::CollisionEvaluatorType::DISCRETE_CONTINUOUS; 
    composite_profile->collision_constraint_config.type = trajopt::CollisionEvaluatorType::DISCRETE_CONTINUOUS;

    trajopt_planner->composite_profiles["FREESPACE"] = composite_profile;
    auto trajopt_generator = std::make_unique<MotionPlannerProcessGenerator>(trajopt_planner);
    trajopt_idx = graph->addNode(std::move(trajopt_generator), GraphTaskflow::NodeType::CONDITIONAL);
```

mpowelson, 2020-09-25:
@Levi-Armstrong Have you used the special_collision_cost?

Levi-Armstrong, 2020-09-28:
I looked through the code but did not see anything obvious. It did work at some point but it has not been tested since all of the changes.

mpowelson, 2020-09-28:
Found it. This was (unsurprisingly) my fault. I had an erroneous profile name and a misunderstanding of what would happen if the profile was not found. 

I was setting a profile called DEFAULT and was not paying attention to what the input was because I thought that DEFAULT would get used if it wasn't found. That is not the case. DEFAULT is only used if the profile is empty - which I'm not even sure is possible since it is a require arg in the constructor. If the profile is not found, it uses the actual default profile (ie default constructor) not whatever you set to DEFAULT.  Personally, I'm not sure if that's how it should work, but I am also not sure that that behavior is uniform across all of our processes. We probably should wrap that into a utility so that it is always the same.

Levi-Armstrong, 2020-09-28:
> Found it. This was (unsurprisingly) my fault. I had an erroneous profile name and a misunderstanding of what would happen if the profile was not found.
> 
> I was setting a profile called DEFAULT and was not paying attention to what the input was because I thought that DEFAULT would get used if it wasn't found. That is not the case. DEFAULT is only used if the profile is empty - which I'm not even sure is possible since it is a require arg in the constructor. If the profile is not found, it uses the actual default profile (ie default constructor) not whatever you set to DEFAULT. Personally, I don't think that's how it should work, but I am also not sure that that behavior is uniform across all of our processes. We probably should wrap that into a utility so that it is always the same.

I agree with the proposed change.

closed 2020-10-28 by mpowelson