tesseract-robotics/tesseract #404: WaypointInCollision doesn't ever return collisions

mpowelson, 2020-10-08:
Somewhere we discussed that [WaypointInCollision](https://github.com/ros-industrial-consortium/tesseract/blob/cf27ce01e06b880e8bcb2e26f95ec04281f157e6/tesseract/tesseract_planning/tesseract_process_managers/src/process_generators/fix_state_collision_process_generator.cpp#L46) doesn't seem to be working. I've again run into this where inputs that pass through this function without problem don't satisfy trajopt contraints. We need to make sure that all of our collision checks are done consistently across, this, the post check processes, and the motion planners.

mpowelson, 2020-10-08:
This seems to work instead.

```

      std::vector<tesseract_collision::ContactResultMap> contacts;

      std::vector<std::string> joint_names;
      std::vector<double> joint_values(start_pos.data(), start_pos.data() + start_pos.size());
      auto state = input.tesseract->getEnvironment()->getState(joint_names, joint_values);

      const bool no_collisions = !tesseract_environment::checkTrajectoryState(contacts, *manager, state);

      {
        for (std::size_t i = 0; i < contacts.size(); i++)
          for (const auto& contact_vec : contacts[i])
            for (const auto& contact : contact_vec.second)
              CONSOLE_BRIDGE_logWarn("%s", ("Links: " + contact.link_names[0] + ", " +
                                       contact.link_names[1] + " Dist: " + std::to_string(contact.distance))
                                          .c_str());
      }
```

Levi-Armstrong, 2020-10-08:
The issue is here. It should not be `traj.rows() - 1` it should just be `traj.rows()`

https://github.com/ros-industrial-consortium/tesseract/blob/cf27ce01e06b880e8bcb2e26f95ec04281f157e6/tesseract/tesseract_environment/include/tesseract_environment/core/utils.h#L345

mpowelson, 2020-10-08:
Yep. That would do it.

mpowelson, 2020-10-08:
I wonder if we are exploiting that bug anywhere.

Levi-Armstrong, 2020-10-08:
Not sure. I looked through that file and did not see anything else.

Levi-Armstrong, 2020-10-09:
@mpowelson Do you have a PR in the works to fix the issue; If not I can create one?

mpowelson, 2020-10-09:
I don't. Unfortunately, I won't get around to doing that this week. Sometime soon though I hope to try to get some of these things fixed rather than just hacked in the branch I am working from. I've got a list of things I'd like to do regarding collisions.

- [ ] Fix this issue
- [ ] Collision checkers segfault if LVS is set to 0
- [ ] Use safety margin in profile in fixStateCollisionProcessGenerator (it is used in trajopt but not in the precheck)
- [ ] Add way to set special contact pairs in post check process

Ultimately, I think we should consider trying to unify our collision checking a bit. For instance the last point, I don't think we have any utilities to that collision check with the link pair specific collision distance like trajopt does. I think it would be good if we could get to a single format for a collision config, so you could use the same config for fixStateCollisionProcessGenerator, motion planners, and the post checks.

mpowelson, 2020-10-09:
Also, I am still tracking an issue in collision checking. I am able to plan into places I can't plan out of (in sim). Sometimes it is just within my safety margin, but sometimes it is actually in collision. I imagine we are missing another case of what you found somewhere and the last point isn't being checked.

mpowelson, 2020-10-09:
@Levi-Armstrong In discrete_contact_check_process_generator.cpp should we be setting the active links based on the ManipulatorInfo? That isn't handled anywhere else is it?

Levi-Armstrong, 2020-10-09:
I believe so.

mpowelson, 2020-11-05:
#426 checked off the last thing in the list, so I will close this

closed 2020-11-05 by mpowelson