tesseract-robotics/tesseract #410: Instructions don't pass profiles to results

mpowelson, 2020-10-15:
I'm not sure where this should best be fixed, but the problem is that the results in the ProcessInput don't get populated with the same profile as the instructions (inputs). This means that any process that uses the results profile (currently only time parameterization I think, but potentially execution) always uses the defaults.

Levi-Armstrong, 2020-10-15:
I would believe it is the responsibility of the seed generator to fill this information out. I just checked the simple planner generators and they are not setting them.

mpowelson, 2020-10-15:
Yeah. I am using `generateSkeletonSeed`. It looks like it sets the Plan/Move Instruction profiles but not the composites.

mpowelson, 2020-10-15:
Nevermind. It looks like it is. Maybe I'm doing something wrong on my side.

mpowelson, 2020-10-21:
To turn this issue into concrete action items.
- [ ] Unit test verifying that generateSkeletonSeed and any other seed generators set the profile

mpowelson, 2020-10-27:
I added a unit test in #431. However, in doing that I realized the real problem. The profile is set correctly in generateSkeletonSeed. However that isn't passed on to the MoveInstructions in the simple planner does not pass the profile from the composite to the individual move instructions.

mpowelson, 2020-10-27:
In fact, now that I am looking at this, it isn't clear to me why we need the generateSkeletonSeed step. The simple planner should recursively handle composite instructions.

To address this issue though, I think I am going to change the step generators to take a PlanInstruction to get meta data from instead of just the manipulator info. Then we can set the profile and description as well.

mpowelson, 2020-10-27:
Actually, base_instructin is already what we want. I'm not sure why the manipulator info is being passed in anyway. There is some logic going to with manip_info.getCombined to combine manipulator infos. Regardless, I should just be able to add `setProfile(base_instruction.getProfile());`

closed 2020-11-02 by mpowelson