tesseract-robotics/tesseract #422: LVS interpolation segfaults on (j2-j1).norm()

mpowelson, 2020-10-21:
This specific case occurs in lvs_interpolation.cpp line 407. To replicate, run the tesseract_process_managers unit tests switching the simple planner to use the LVS profile in simple_motion_planner.cpp (line 75). After #412 is merged, change

```
plan_profiles["DEFAULT"] = std::make_shared<SimplePlannerDefaultPlanProfile>();
```
to
```
plan_profiles["DEFAULT"] = std::make_shared<SimplePlannerLVSDefaultPlanProfile>();
```

Levi-Armstrong, 2020-11-12:
@mpowelson Has this been resolved with the latest changes to LVS?

mpowelson, 2020-11-12:
I'm not sure. Like I said though, if you change the simple planner to use LVS instead, the tests should start failing. So if you are using it, then I'd guess you probably fixed it.

Levi-Armstrong, 2020-11-12:
Yea it is the default now and everything is working. I will close, please reopen if you try it again and experience issues.

closed 2020-11-12 by Levi-Armstrong