tesseract-robotics/tesseract #427: Add generic collision config that all collision processes use.

mpowelson, 2020-10-23:
These should all use the same config type for defining contact distance, special collision pairs, etc.

- [ ] ContinuousContactCheckProcessGenerator
- [ ] DiscreteContactCheckProcessGenerator
- [ ] FixStateCollisionProcessGenerator
- [ ] Motion planners

mpowelson, 2020-10-28:
@Levi-Armstrong Do we still want the coeff or is just the distance sufficient here. I'm thinking that is fine. 
So instead of this from trajopt
```
  /// A map of link pair to contact distance setting [dist_pen, coeff]
  tesseract_common::AlignedUnorderedMap<std::string, Eigen::Vector2d> pair_lookup_table_;
```
It will be this 
```
  /// A map of link pair to contact distance setting 
  std::unordered_map<std::string, double> pair_lookup_table_;
```

mpowelson, 2020-11-16:
> @Levi-Armstrong Do we still want the coeff or is just the distance sufficient here. I'm thinking that is fine.
> So instead of this from trajopt
>
> ```
>   /// A map of link pair to contact distance [...]

For reference, we eliminated the coeff here.

mpowelson, 2020-11-16:
What do we want in this datastructure? Right now I have (I'll keep this post updated)

```
struct CollisionCheckConfig
{
  CollisionMarginData collision_margin_data;
  double longest_valid_segment_length_{ 0.005 };

};
```

mpowelson, 2020-11-17:
@Levi-Armstrong What would you think about me adding a CollisionEvaluatorType in Tesseract sort of like we have in trajopt (it would probably replace it when we refactor into using trajopt_ifopt). It would replace the enable_lvs bool above. Something like this.
```
enum class CollisionEvaluatorType
{
  DISCRETE,
  LVS_DISCRETE,
  CONTINUOUS,
  LVS_CONTINUOUS
};
```
The issue here of course would be if someone specifies CONTINUOUS but passes in a discrete contact manager to a utility. However, I think in those cases we would just print a warning an use the equivalent for that manager. Things like planners that have access to both managers need to know which to use.

mpowelson, 2020-11-17:
Which makes me think, should I add a string for selecting the contact manager?

Levi-Armstrong, 2020-11-17:
> For reference, we eliminated the coeff here.

So I don't believe I have every change change the coefficient on a link by link basis, so maybe we just have a single coefficient per collision cost/constraint?

Levi-Armstrong, 2020-11-17:
> What do we want in this datastructure? Right now I have (I'll keep this post updated)
> 
> ```
> struct CollisionCheckConfig
> {
>   CollisionMarginData collision_margin_data;
>   tesseract_collision::ContactRequest contact_request{ tesseract_collision::ContactTestType::FIRST };
>   tesseract_common::Verbosity verbosity{ tesseract_common::Verbosity::INFO };
>   bool enable_lvs{ false };
>   double longest_valid_segment_length{ 0.005 };
> };
> ```

I like the addition of the CollisionEvaluatorType; would this replace the enable_lvs in the above data structure? I noticed the verbosity member and I have been thinking of something similar. I originally thought console bridge logger level was a global setting but I believe it is by process, so was wondering if we should remove passing in verbosity through the API and just use setting the logger level. I believe I have seen you do this in other places; what do you think?

Levi-Armstrong, 2020-11-17:
> The issue here of course would be if someone specifies CONTINUOUS but passes in a discrete contact manager to a utility. However, I think in those cases we would just print a warning an use the [...]

Yea, I think we would just throw an exception if this happens and they will need to fix it.

mpowelson, 2020-11-17:
> So I don't believe I have every changed the coefficient on a link by link basis, so maybe we just have a single coefficient per collision cost/constraint?

I'll have to look at how much work it would be to keep this functionality. I imagine it is not something most people would do like you said. Regardless, I took it out of the generic data structure. TrajOpt will likely still have a separate datastructure that either inherits from this one or contains one of these and adds some information like coeff, use_weighted_sum, etc.

> was wondering if we should remove passing in verbosity through the API and just use setting the logger level.

That would probably make sense in most cases including this one. I can remove the verbose flag and just use console_bridge logger level.

> Yea, I think we would just throw an exception if this happens and they will need to fix it.

Ok. I can do that instead.

Levi-Armstrong, 2020-11-17:
Sounds goods.

PR #496 (2021-01-05): Update planners to use CollisionCheckConfig
I did not update trajopt because it has its own data structure. While we could go through and update it, I'd rather just focus on getting the ifopt planner working.

Similarly I added the ContactMarginData to OMPL but did not use the CollisionCheckConfig since OMPL handles LVS itself. I thought it would be confusing to have 2 longest valid segment terms where only one gets used.

Closes #427
closed 2021-01-05 by mpowelson