tesseract-robotics/tesseract #429: How to add toleranced joint position using TCL?

mpowelson, 2020-10-26:
I don't think it is possible at the moment. The question to me is do we 

1) Add a TolerancedJointWaypoint
2) Add optional tolerances to JointWaypoint
3) Handle this in a profile

My use case is a freespace trajectory like this {joint_cnt, toleranced via point, ..., toleranced via point, joint_cnt}. My preference I think would be 1 or maybe 2 so that the trajectory could still all use the same profile and tolerances can be set at request time.

Levi-Armstrong, 2020-10-27:
I believe the original plan was for the profile to handle things like this. I think either way you have to either change the waypoint or the profile. I also think it can get confusing if we start adding planner specific information in the waypoint versus using the profiles.  Do you have more details on the use case? I could see an issues with using the profile approach if the tolerance is constantly changing, but maybe this could be build into the profiles.

mpowelson, 2020-10-27:
Yeah. I get not wanting to add too much to the waypoints. However, I also wouldn't necessarily call tolerances planner specific. Some planners may not support tolerances, but it is still specific to the task at hand rather than the solution method.

Regardless of how we implement it, this does demonstrate a bit of a limitation in the profile system in that to change a parameter at execution time you have to have access to both the process manager and the process input. An example I am currently using is time parameterization. I want the ability to change the velocity scaling factor at runtime, so I have to keep a copy of the IterativeSplineParameterizationProfile::Ptr around so I can change it.

Going back to the tolerances, an application I can think of is something like a pick and place. Perhaps the tolerance on the place position depends on a perception system that tells how cluttered the place location is. It's not impossible to do with the tolerances being specified in the profile. You just have to modify the planning server in order to set it rather than adjusting the waypoint.

Levi-Armstrong, 2020-10-28:
Sounds good, lets add tolerances to the CartesianWaypoint and JointWaypoint or should we create new waypoint Types? 

The second item, since the profiles can also be serialized what about adding support for adding profiles to the planning request?

mpowelson, 2020-10-29:
I don't really feel strongly about whether or not they should be their own types. Adding it to the existing ones would probably be easiest.

Yeah I was thinking it might be nice to have some way to overwrite the profiles. I haven't given it too much thought as to where that should go.

mpowelson, 2021-01-07:
Addressed in #443

closed 2021-01-07 by mpowelson