tesseract-robotics/tesseract #432: Erroneous clang-tidy errors when building tesseract_collision

mpowelson, 2020-10-27:
I mentioned this on a pull request but want to open an issue so we will have a record of the solution if/when we figure out what is going on. When building tesseract_collision with `-DTESSERACT_ENABLE_CLANG_TIDY=ON -DTESSERACT_ENABLE_TESTING=ON` I get a bunch of misc-unused-parameters warnings that cause the build to fail. They are all in headers like bullet_utils.h, common.h, types.h, and fcl_utils.h An example:
```
/src/tesseract/tesseract/tesseract_collision/include/tesseract_collision/fcl/fcl_utils.h:215:99: error: parameter 'dynamic_manager' is unused [misc-unused-parameters,-warnings-as-errors]
                                         const std::unique_ptr<fcl::BroadPhaseCollisionManagerd>& dynamic_manager)
```

Additionally (possibly unrelated), I get a clang-diagnostic-error that `'omp.h' file is not found` in /usr/include/eigen3/Eigen/Core:247:10

mpowelson, 2020-10-27:
I just remembered the search feature and realized that this is probably a duplicate of #198

Levi-Armstrong, 2020-10-27:
Could you try removing the target_clang_tidy and target_code_coverage for library `${PROJECT_NAME}_core ${PROJECT_NAME}_test_suite and ${PROJECT_NAME}_test_suite_benchmarks` in tesseract_collision?  Now that I am looking at it these probably should not be added to interface targets.

mpowelson, 2020-10-27:
Ok. I'll try that. For the record. Because of #198 I already have clang-6 installed and the symlink was fine. So that didn't fix it.

mpowelson, 2020-10-27:
That didn't seem to help

mpowelson, 2020-10-27:
I just reinstalled OpenMP and it seems to work now.  Installing with apt said something was broken. When I ran `sudo aptitude install libomp-dev` I got this

```

     Remove the following packages:                                             
1)     libomp-8-dev [1:8-3~ubuntu18.04.2 (bionic-security, bionic-updates, now)]
2)     libomp5-8 [1:8-3~ubuntu18.04.2 (bionic-security, bionic-updates, now)]   

     Leave the following dependencies unresolved:                               
3)     clang-8 recommends libomp-8-dev        
```
I followed the recommendation and it seems to work. We'll see if something else pops up broken later.

closed 2020-10-28 by mpowelson