tesseract-robotics/tesseract #435: DiscreteContactCheckProcessGenerator returning 0 when contact distance > 0

mpowelson, 2020-10-29:
@Levi-Armstrong I am getting some processes that have DiscreteContactCheckProcessGenerator return 0 but when the contact distance prints out, it is > 0. The contact checkers shouldn't be returning "collisions" with distances greater than 0 should they? Or is that distance before inflation?

Levi-Armstrong, 2020-10-29:
That is the actual distance between the two shapes.

mpowelson, 2020-10-29:
Ok. So as long as that number is smaller than 2x the safety margin, that is correct?

Levi-Armstrong, 2020-10-29:
It should be smaller than 1x the safety margin because the safety margin (contact distance threshold is the relative distance). I also think we should use a different name that safety margin because you could use negative numbers, maybe ContactThresholdData?

The current approach if you set a contact distance threshold of 0.1m it will expand all shapes bounding box by 0.05m. 

Example:
global contact distance threshold: 0.1m
case 1: link pair(link_1, link_2) = 0.2; which is greater than global then it should use 0.1m
case 2: link pair(link_1, link_2) = 0.05; which is less than global then it should also use 0.1m so it still catches collisions with the other links due to the global contact distance.

Now that I walked through it, I think you will still do everything as before using the global contact threshold, you would just use the link pair data to decide if it should be logged as a contact violation. Does this make sense?

mpowelson, 2020-10-29:
contact_distance is what is used inside the contact managers. SafetyMargin is what was in TrajOpt.

Levi-Armstrong, 2020-10-29:
> contact_distance is what is used inside the contact managers. SafetyMargin is what was in TrajOpt.

Yea, I think that was a poor name choice on my part. When I see safety margin I automatic assume the parts should not be in contact which may not always be the case.

mpowelson, 2020-10-30:
@Levi-Armstrong What prompted this was me chasing an issue with inconsistency between `FixStateCollisionProcessGenerator` and `DiscreteContactCheckProcessGenerator`. I'm getting plans where FixStateCollisionProcessGenerator  with distance set to 0.005 will say it is not in collision, but then a DiscreteContactCheckProcessGenerator set to 0.004 right after that says it has a distance of 0.003907 and will fail because it is in collision.

Levi-Armstrong, 2020-10-30:
I think your at the numerical limits of the convex to convex algorithm. If you increase the fix start state to 0.008 does it go away? Also FCL's convex to convex seems to be more accurate at the range of 0.001.

mpowelson, 2020-10-30:
Even with the double precision bullet? It's only very occasionally, so it is hard to replicate. I could probably increase it a little bit, but parts of the links do get pretty close, so it can't be 0.02 for instance.

mpowelson, 2020-10-30:
If that really is the case, should we add the ability to change the units used in Bullet? Ie, like scale everything up by 10x or something? 5mm doesn't seem that small.

closed 2020-11-16 by mpowelson