tesseract-robotics/tesseract #445: JointLimits grow over time

mpowelson, 2020-11-13:
@Levi-Armstrong Using our new change joint limits code I am getting errors like this. It seems every time we call it, the limits matrix in the forward kinematics object grows by n_dof

```
[ WARN] ros.rosconsole_bridge.console_bridge: Invalid limits when clamping Waypoint. Waypoint size: 6, Limits size: 30
[ WARN] ros.rosconsole_bridge.console_bridge: Invalid limits when clamping Waypoint. Waypoint size: 6, Limits size: 30
```

mpowelson, 2020-11-13:
KDL setLimits looks fine to me. I'm thinking it must be upstream of there.

Levi-Armstrong, 2020-11-13:
Are you applying it using the command?

mpowelson, 2020-11-13:
I'm not doing it via ROS. I'm using `tesseract->getEnvironment()->changeJointPositionLimits(const std::string& joint_name, double lower, double upper)`

Levi-Armstrong, 2020-11-13:
I see the issue. In the kinematics init method it should be clearing out the internal data member.. So when we call init in the update the data is getting appended.

mpowelson, 2020-11-13:
Ok. Well I just wrote a unit test that fails. I'll change that and make sure it passes now.

mpowelson, 2020-11-13:
Do you want me to just do kdl_data_ = KDLChainData() or should we add a clear method?

Levi-Armstrong, 2020-11-13:
I think that should do it. Everything else looks to get overridden.  I am also good with the clear method.

mpowelson, 2020-11-13:
Ok. That did seem to fix it. I'll probably just leave it to that for now.

Levi-Armstrong, 2020-11-13:
Addressed by #446

closed 2020-11-13 by Levi-Armstrong