tesseract-robotics/tesseract #459: OFKT state solver does not update joint limits

mpowelson, 2020-12-01:
Using `Tesseract->getEnvironment()->changeJointPositionLimits()` the limits get changed in the scene graph and it looks like in the limits_ MatrixX2d in the solver. However, the nodes_ store their own limits and those do not get updated.

mpowelson, 2020-12-01:
@Levi-Armstrong How do you think we should fix this? I kind of think we should just remove limits_ from the revolute node. None of the other joint types have them, and I kind of feel that joint limits should be handled higher up in the environment and the solver should just solve whatever it is given.

Levi-Armstrong, 2020-12-01:
I do not have issue with the change, though not sure what the impact is with the change since it is at a low level.

Levi-Armstrong, 2020-12-01:
Wait are you talking about removing it from the OFKT Revolute Node? If so that should not be an issue.

mpowelson, 2020-12-01:
Yes. That's what I was wanting to do. All it is used for is clamping the inputs to joint limits.

Levi-Armstrong, 2020-12-01:
I am good with you removing it.

closed 2020-12-03 by mpowelson