tesseract-robotics/tesseract #464: -Werror=sign-conversion errors

johnwason, 2020-12-08:
I am getting a bunch of errors in unit tests due to sign conversion errors, for example:

```
tesseract/tesseract/tesseract_collision/include/tesseract_collision/test_suite/collision_box_cylinder_unit.hpp:148:55: error: conversion to ‘std::array<Eigen::Matrix<double, 3, 1>, 2>::size_type’ {aka ‘long unsigned int’} from ‘__gnu_cxx::__alloc_traits<std::allocator<int>, int>::value_type’ {aka ‘int’} may change the sign of the result [-Werror=sign-conversion]
  148 |     EXPECT_NEAR(result_vector[0].nearest_points[idx[0]][0], 0.5, 0.001);
      |                                     
```

I am seeing these errors on my feature/CommandLanguagePython branch, but not on the base feature/CommandLanguage branch. I am not sure why this is occurring. The link to the branch is here: https://github.com/johnwason/tesseract/tree/feature/CommandLanguagePython
tesseract_python isn't building yet, so add `CATKIN_IGNORE` to that directory for now. I am using Ubuntu 20.04 x64 with ROS Noetic.

Levi-Armstrong, 2020-12-09:
For lines 148  try adding a static_cast<std::size_t>(idx[0]) and 149 static_cast<std::size_t>(idx[1]) to see if the error goes away here.

```
 EXPECT_NEAR(result_vector[0].nearest_points[idx[0]][0], 0.5, 0.001);
 EXPECT_NEAR(result_vector[0].nearest_points[idx[1]][0], -0.05, 0.001);
```

Levi-Armstrong, 2020-12-11:
The issue was that c-arrays were switched to std::array triggering the `-Werror=sign-conversion`

closed 2020-12-11 by Levi-Armstrong