tesseract-robotics/tesseract #469: Consider adding a .clang-tidy file

mpowelson, 2020-12-10:
We could potentially add a file to use as defaults rather than listing it in tesseract_common. I temporarily added "-dump-config" to the cmake macro to generate a config from what we are currently setting. It is below

```
---
Checks:          'clang-diagnostic-*,clang-analyzer-*,-*,clang-analyzer-*,bugprone-*,cppcoreguidelines-avoid-goto,cppcoreguidelines-c-copy-assignment-signature,cppcoreguidelines-interfaces-global-init,cppcoreguidelines-narrowing-conversions,cppcoreguidelines-no-malloc,cppcoreguidelines-slicing,cppcoreguidelines-special-member-functions,misc-*,modernize-*,performance-*,readability-avoid-const-params-in-decls,readability-container-size-empty,readability-delete-null-pointer,readability-deleted-default,readability-else-after-return,readability-function-size,readability-identifier-naming,readability-inconsistent-declaration-parameter-name,readability-misleading-indentation,readability-misplaced-array-index,readability-non-const-parameter,readability-redundant-*,readability-simplify-*,readability-static-*,readability-string-compare,readability-uniqueptr-delete-release,readability-rary-objects'
WarningsAsErrors: ''
HeaderFilterRegex: '.*'
AnalyzeTemporaryDtors: false
FormatStyle:     none
User:            matthew
CheckOptions:    
  - key:             bugprone-argument-comment.StrictMode
    value:           '0'
  - key:             bugprone-assert-side-effect.AssertMacros
    value:           assert
  - key:             bugprone-assert-side-effect.CheckFunctionCalls
    value:           '0'
  - key:             bugprone-dangling-handle.HandleClasses
    value:           'std::basic_string_view;std::experimental::basic_string_view'
  - key:             bugprone-string-constructor.LargeLengthThreshold
    value:           '8388608'
  - key:             bugprone-string-constructor.WarnOnLargeLength
    value:           '1'
  - key:             cppcoreguidelines-no-malloc.Allocations
    value:           '::malloc;::calloc'
  - key:             cppcoreguidelines-no-malloc.Deallocations
    value:           '::free'
  - key:             cppcoreguidelines-no-malloc.Reallocations
    value:           '::realloc'
  - key:             cppcoreguidelines-special-member-functions.AllowMissingMoveFunctions
    value:           '0'
  - key:             cppcoreguidelines-special-member-functions.AllowSoleDefaultDtor
    value:           '0'
  - key:             google-readability-braces-around-statements.ShortStatementLines
    value:           '1'
  - key:             google-readability-function-size.StatementThreshold
    value:           '800'
  - key:             google-readability-namespace-comments.ShortNamespaceLines
    value:           '10'
  - key:             google-readability-namespace-comments.SpacesBeforeComments
    value:           '2'
  - key:             misc-definitions-in-headers.HeaderFileExtensions
    value:           ',h,hh,hpp,hxx'
  - key:             misc-definitions-in-headers.UseHeaderFileExtension
    value:           '1'
  - key:             misc-misplaced-widening-cast.CheckImplicitCasts
    value:           '0'
  - key:             misc-sizeof-expression.WarnOnSizeOfCompareToConstant
    value:           '1'
  - key:             misc-sizeof-expression.WarnOnSizeOfConstant
    value:           '1'
  - key:             misc-sizeof-expression.WarnOnSizeOfThis
    value:           '1'
  - key:             misc-suspicious-enum-usage.StrictMode
    value:           '0'
  - key:             misc-suspicious-missing-comma.MaxConcatenatedTokens
    value:           '5'
  - key:             misc-suspicious-missing-comma.RatioThreshold
    value:           '0.200000'
  - key:             misc-suspicious-missing-comma.SizeThreshold
    value:           '5'
  - key:             misc-suspicious-string-compare.StringCompareLikeFunctions
    value:           ''
  - key:             misc-suspicious-string-compare.WarnOnImplicitComparison
    value:           '1'
  - key:             misc-suspicious-string-compare.WarnOnLogicalNotComparison
    value:           '0'
  - key:             misc-throw-by-value-catch-by-reference.CheckThrowTemporaries
    value:           '1'
  - key:             modernize-loop-convert.MaxCopySize
    value:           '16'
  - key:             modernize-loop-convert.MinConfidence
    value:           reasonable
  - key:             modernize-loop-convert.NamingStyle
    value:           CamelCase
  - key:             modernize-make-shared.IgnoreMacros
    value:           '1'
  - key:             modernize-make-shared.IncludeStyle
    value:           '0'
  - key:             modernize-make-shared.MakeSmartPtrFunction
    value:           'std::make_shared'
  - key:             modernize-make-shared.MakeSmartPtrFunctionHeader
    value:           memory
  - key:             modernize-make-unique.IgnoreMacros
    value:           '1'
  - key:             modernize-make-unique.IncludeStyle
    value:           '0'
  - key:             modernize-make-unique.MakeSmartPtrFunction
    value:           'std::make_unique'
  - key:             modernize-make-unique.MakeSmartPtrFunctionHeader
    value:           memory
  - key:             modernize-pass-by-value.IncludeStyle
    value:           llvm
  - key:             modernize-pass-by-value.ValuesOnly
    value:           '0'
  - key:             modernize-raw-string-literal.ReplaceShorterLiterals
    value:           '0'
  - key:             modernize-replace-auto-ptr.IncludeStyle
    value:           llvm
  - key:             modernize-replace-random-shuffle.IncludeStyle
    value:           llvm
  - key:             modernize-use-auto.RemoveStars
    value:           '0'
  - key:             modernize-use-default-member-init.IgnoreMacros
    value:           '1'
  - key:             modernize-use-default-member-init.UseAssignment
    value:           '0'
  - key:             modernize-use-emplace.ContainersWithPushBack
    value:           '::std::vector;::std::list;::std::deque'
  - key:             modernize-use-emplace.SmartPointers
    value:           '::std::shared_ptr;::std::unique_ptr;::std::auto_ptr;::std::weak_ptr'
  - key:             modernize-use-emplace.TupleMakeFunctions
    value:           '::std::make_pair;::std::make_tuple'
  - key:             modernize-use-emplace.TupleTypes
    value:           '::std::pair;::std::tuple'
  - key:             modernize-use-equals-default.IgnoreMacros
    value:           '1'
  - key:             modernize-use-noexcept.ReplacementString
    value:           ''
  - key:             modernize-use-noexcept.UseNoexceptFalse
    value:           '1'
  - key:             modernize-use-nullptr.NullMacros
    value:           'NULL'
  - key:             modernize-use-transparent-functors.SafeMode
    value:           '0'
  - key:             modernize-use-using.IgnoreMacros
    value:           '1'
  - key:             performance-faster-string-find.StringLikeClasses
    value:           'std::basic_string'
  - key:             performance-for-range-copy.WarnOnAllAutoCopies
    value:           '0'
  - key:             performance-inefficient-string-concatenation.StrictMode
    value:           '0'
  - key:             performance-inefficient-vector-operation.VectorLikeClasses
    value:           '::std::vector'
  - key:             performance-move-const-arg.CheckTriviallyCopyableMove
    value:           '1'
  - key:             performance-move-constructor-init.IncludeStyle
    value:           llvm
  - key:             performance-type-promotion-in-math-fn.IncludeStyle
    value:           llvm
  - key:             performance-unnecessary-value-param.IncludeStyle
    value:           llvm
  - key:             readability-function-size.BranchThreshold
    value:           '4294967295'
  - key:             readability-function-size.LineThreshold
    value:           '4294967295'
  - key:             readability-function-size.NestingThreshold
    value:           '4294967295'
  - key:             readability-function-size.ParameterThreshold
    value:           '4294967295'
  - key:             readability-function-size.StatementThreshold
    value:           '800'
  - key:             readability-identifier-naming.IgnoreFailedSplit
    value:           '0'
  - key:             readability-simplify-boolean-expr.ChainedConditionalAssignment
    value:           '0'
  - key:             readability-simplify-boolean-expr.ChainedConditionalReturn
    value:           '0'
  - key:             readability-static-accessed-through-instance.NameSpecifierNestingThreshold
    value:           '3'
...
```
As an additional reference, [here](https://github.com/ros-planning/moveit/blob/master/.clang-tidy) is MoveIts config.

Levi-Armstrong, 2020-12-30:
It looks like this should work but it need the .clang-tidy for each package at the same level of the CMakeLists.txt for the package. We could create one for the repository and symbolically link it to each package. Though I have not tested this.

mpowelson, 2020-12-30:
Would we then be able to use the cmake_common_scripts macro? Since we wouldn't have to define our args in the cmake?

mpowelson, 2020-12-30:
I guess we wouldn't be able to only define some things as errors when running tests

Levi-Armstrong, 2020-12-30:
Ah, I did not think of that. I will see if we can pass the file path then we can have a config for normal building and one for tests.

Levi-Armstrong, 2020-12-30:
Though if it is required to pass CI then it may make sense to have them always be errors.

Levi-Armstrong, 2020-12-30:
What do you think about making them always errors?

mpowelson, 2020-12-30:
I think that is probably fine. To be honest, I don't think I knew that they were only errors when you build the tests until I looked at the macro.

Levi-Armstrong, 2020-12-30:
Yea, I think when it was originally setup there was not a way to disable clang-tidy but now that you can define `TESSERACT_ENABLE_CLANG_TIDY=ON/OFF` it does not seem necessary.

mpowelson, 2020-12-31:
Looks like you [can specify a config](https://releases.llvm.org/4.0.0/tools/clang/tools/extra/docs/clang-tidy/index.html)

```
  -config=<string>             -
                                 Specifies a configuration in YAML/JSON format:
                                   -config="{Checks: '*',
                                             CheckOptions: [{key: x,
                                                             value: y}]}"
                                 When the value is empty, clang-tidy will
                                 attempt to find a file named .clang-tidy for
                                 each source file in its parent directories.
```

mpowelson, 2020-12-31:
Here is the new file after the changes for clang-tidy 10. I could not get it to find the file though. I tried putting it in the root of the repo and in each individual package. Neither seemed to work. If you want to take a look, go for it, but at this point I'm thinking we should just stick with what we have.
[tesseract.clang-tidy.txt](https://github.com/ros-industrial-consortium/tesseract/files/5756216/tesseract.clang-tidy.txt)

mpowelson, 2021-02-18:
I'm going to close this since what we have works, and I don't think anyone is planning to work on this.

closed 2021-02-18 by mpowelson