tesseract-robotics/tesseract #491: Add joint angle seed or joint configuration seed to CartesianWaypoint

johnwason, 2020-12-23:
Currently the planners will randomly select a joint configuration when inverse kinematics are used for CartesianWaypoint. For 6 axis robots, there are 8 configurations. For higher joint numbers, the possible configurations are infinite. ABB uses "configuration" information. This document has a good explanation:

https://fzi-robot-interface-proposal.readthedocs.io/en/latest/industry/ABB.html

(I haven't seen this proposal before, I need to reed through it)

We should include either a "joint seed" position or a "joint configuration" parameter so that the chosen configuration can be specified.

johnwason, 2020-12-23:
https://fzi-robot-interface-proposal.readthedocs.io/en/latest/proposal/index.html

This proposal adds a "CartesianPosture" type:

```
CartesianPosture posture
      string [] posture_joint_names
      float64[] posture_joint_values
```

Levi-Armstrong, 2020-12-28:
The way we have been supporting this is within the profile. In tesseract_motion_planners we have a robot_config with will give you the robot configuration provided the joint names and joint values. This is used in the is state valid for ompl and descartes where either the configuration is excluded or different costs are assigned to wait the sampled states. 

Though this would be useful if you wanted to use the command language for OLP where you record a Cartesian position and need to store the robot configuration. I believe previously we discussed created CartesianStateWaypoint which would be similar to what you have proposed. Is this the use case you had in mind?

johnwason, 2020-12-29:
I don't know that a profile is the appropriate way to handle this, since the user may want to specify the configuration for each waypoint rather than for the entire program. Would a CartesianStateWaypoint be an input or an output of the planner? I am thinking of needing to specify the configuration for the input.

There may be a need to have a way to pass arbitrary parameters as part of the instruction. Maybe something like a `map<string,vector<double> >` to pass parameters we haven't thought of yet, or something more generic like `map<string,std::any>` although I am not as familiar with `std::any` so I am not sure how that would work.

Levi-Armstrong, 2020-12-29:
> I don't know that a profile is the appropriate way to handle this, since the user may want to specify the configuration for each waypoint rather than for the entire program. Would a [...]

I guess it depends on what you are trying to accomplish with this. Is this intended to be used during planning or executions?

johnwason, 2020-12-29:
We will need to allow the user to pass a desired configuration during the planning request, like the example ABB method.

Levi-Armstrong, 2020-12-29:
If it is for planning then the recommended place is in the Plan Profile which is specific to each waypoint, where the composite profile is related to a collection of waypoints. I can provide an example on how are leveraging the profiles to control robot configuration on an ABB. Usually in our use cases we do not confine it to a single configuration but to a subset of configurations. If it is a Cartesian pose with a single configuration then I would view this as a fixed joint pose given a 6 DOF ABB robot. Is this what you want to accomplish without having to solve IK and get the joint solution?

Levi-Armstrong, 2022-01-26:
This has been added.

closed 2022-01-26 by Levi-Armstrong