tesseract-robotics/tesseract #504: addJoint() in Environment

johnwason, 2021-01-04:
`addJoint()` is implemented in the scene graph, but not exposed to `Environment`. It looks like the idea is to have joints be added with links. This is a safer design, but prevents the creation of closed kinematic loops. Is this an intentional design choice?

Levi-Armstrong, 2021-01-04:
Yea, it was intentional until a state solver was added that supports closed kinematic loops. This was the thought around adding direct support for subgraph where each subgraph would be equivalent to a model which could have its own state solver. Do you a current use case?

johnwason, 2021-01-04:
I am working on the Robot Raconteur tesseract server and noticed it was missing. I don't think that most planners support closed loops, so I will probably leave off the function. I wanted to make sure.

closed 2021-01-04 by johnwason