tesseract-robotics/tesseract #510: Unable to access ProcessInfoMap through planning server

mpowelson, 2021-01-06:
This is in the ProcessInput but is not in the ProcessPlanningFuture. I'm thinking we should just add it to the future. It looks like

```
std::map<std::size_t, ProcessInfo::ConstPtr>
```

mpowelson, 2021-01-06:
@Levi-Armstrong

Levi-Armstrong, 2021-01-06:
Sound good.

mpowelson, 2021-01-06:
Well it isn't going to be that simple. The map is stored in a threadsafe container in the ProcessInput as a shared_ptr. I suppose we can pass the container into ProcessInput when we construct it and then add a pointer to it to the ProcessPlanningFuture as well. Maybe it should go in the ProcessInterface? Then I could add the getters as well, so users don't have to know about the container.

On a slightly different topic, ProcessInterface is sort of mixing terminology here. It really is a taskflow interface since you can't cancel a single process.

Levi-Armstrong, 2021-01-06:
> On a slightly different topic, ProcessInterface is sort of mixing terminology here. It really is a taskflow interface since you can't cancel a single process.

I agree based in the process generator versus taskflow generator. I am good with a rename. Might as well do it now. Along those same lines should ProcessInput be renamed to TaskflowInput? Also should we rename anything else in the core folder? I just prefixed everything with process without really thinking about it.

mpowelson, 2021-01-06:
We should probably open another issue to figure out what everything should be called. ProcessGenerators should probably be TaskGenerators. Then we would have TaskGenerators, TaskflowGenerators, and the ProcessPlanningServer + accessories.

mpowelson, 2021-01-06:
I can submit a PR tomorrow to address this issue. The main thing we need to decide though is do we want it in the ProcessInterface or the ProcessPlanningFuture directly?

Levi-Armstrong, 2021-01-06:
I don't have particular preference, but if it is added to the interface then we should update the documentation. It does seem like the simplest solution.

Levi-Armstrong, 2021-01-07:
Addressed in #514

closed 2021-01-07 by Levi-Armstrong