tesseract-robotics/tesseract #511: Address Naming in Tesseract Process Managers

mpowelson, 2021-01-06:
The naming of some classes don't make a lot of sense as they are fragments of legacy code.

"We should probably open another issue to figure out what everything should be called. ProcessGenerators should probably be TaskGenerators. Then we would have TaskGenerators, TaskflowGenerators, and the ProcessPlanningServer + accessories."

_Originally posted by @mpowelson in https://github.com/ros-industrial-consortium/tesseract/issues/510#issuecomment-755022459_

Levi-Armstrong, 2021-01-06:
Current Idea:

ProcessGenerator -> TaskGenerator
ProcessInterface -> TaskflowInterface

Are there any other to be renamed?

Levi-Armstrong, 2021-01-06:
Would we also rename?

ProcessInfo -> TaskInfo
ProcessInput -> TaskflowInput

mpowelson, 2021-01-06:
I'd call ProcessInput TaskInput (rather than taskflow input) since we divide it up into sub ProcessInputs inside the taskflow generator before sending it to each Task.

I think ProcessInteface, ProcessPlanningFuture, ProcessPlanningRequest, and ProcessPlanningServer all all fine since they refer to a robotic process.

So we have things that refer to

- Taskflow Tasks
- Taskflow Taskflows
- Robotic processes (ie applications)

Levi-Armstrong, 2021-01-06:
Sounds good

mpowelson, 2021-01-06:
@Levi-Armstrong Are you addressing this or would you like me to?

Levi-Armstrong, 2021-01-06:
If you have time that would be great, if not I may be able to get later this week. I am currently working on releasing the dependencies.

mpowelson, 2021-01-06:
Summary
- ProcessGenerator -> TaskGenerator
- ProcessInterface -> TaskflowInterface
- ProcessInfo -> TaskInfo
- ProcessInfoContainer -> TaskInfoContainer
- ProcessInput -> TaskInput

Levi-Armstrong, 2021-01-07:
This has been addressed in #514

closed 2021-01-07 by Levi-Armstrong