tesseract-robotics/tesseract #521: ROS Buildfarm Release "Noetic"

Levi-Armstrong, 2021-01-11:
- [ ] cmake_common_scripts
- [ ] opw_kinematics
- [ ] descartes_light
- [ ] osqp
- [ ] osqp_eigen
- [ ] qpOASES
- [ ] tesseract
- [ ] tesseract_planning
- [ ] tesseract_ros
- [ ] tesseract_python

Levi-Armstrong, 2021-04-08:
@mpowelson, @marip8 and @johnwason I believe this repository is ready to be released. Any objections to me releasing this repository for Noetic with tag 1.0.0?

johnwason, 2021-04-08:
I am going to try a conda build tonight, let's make sure it will work there.

johnwason, 2021-04-08:
No, it isn't building using conda on windows. I can't look into this until the week after next most likely. We can add patches later to the conda package if needed, or wait until the conda build is working.

marip8, 2021-04-08:
I'm on board for the release. As far as I can tell it looks like the major discussions about de-coupling the planning/visualization code have been addressed and there aren't any outstanding PRs with major changes.

What is your tagging strategy? Are the 0.x.x tags supposed to represent pre-release states and 1+.x.x represent post-release states? Just wondering if the next tag should be 0.3.0 since there really hasn't been a "major" change since the last tag

gavanderhoorn, 2021-04-08:
> What is your tagging strategy? Are the 0.x.x tags supposed to represent pre-release states and 1+.x.x represent post-release states? Just wondering if the next tag should be 0.3.0 since there really hasn't been a "major" change since the last tag

Going to `1.x` immediately may indeed not be what you'd want.

If you're going to use [semver](https://semver.org), seeing a `1.x` release implies quite a few things.

mpowelson, 2021-04-08:
I think I agree that a 0.x release might be a better option right now. Then if that goes well we can plan on a 1.0 release sometime soon.

Levi-Armstrong, 2021-04-08:
@gavanderhoorn, @mpowelson and @marip8 Thank you all for the input. I will begin the release process and use tag 0.4.0.

Levi-Armstrong, 2021-04-10:
Just submitted the [release](https://github.com/ros/rosdistro/pull/29068) on the ROS buildfarm.

Levi-Armstrong, 2021-04-14:
@johnwason Everything has been upgraded to taskflow v3.0.0.

johnwason, 2021-04-14:
@Levi-Armstrong ok I'll give the conda build another try

Levi-Armstrong, 2021-04-23:
The release status can be tacked [here](http://repositories.ros.org/status_page/ros_noetic_default.html?q=tesseract).