tesseract-robotics/tesseract #540: addSceneGraph does not add links to collision

haudren, 2021-02-03:
I was recently in the process of upgrading to the command language version of Tesseract and hit a problem with it: merging another scene graph into my environment resulted in links being correctly added to the environment, but not to the collision checking, and my robot would happily go through walls...

My investigation shows that:
- `addSceneGraph` mostly calls `scene_graph_->insertSceneGraph`, which calls its own `addLink`
- `Environment::addLink` correctly inserts both a link and the collisions, but obviously `SceneGraph::addLink` doesn't, as it does not have access to the managers.
- Because `getDiscreteContactManager` returns a clone of the manager, you can't add externally the collisions

Workaround is thus to call `Environment::registerDefaultContactManagers` to re-generate the list of collision objects. But this feels like an awkward API:
- I call `Environment::addSceneGraph`, I expect it to register itself properly in all nooks and crannies
- As in #539 , it feels quite counter-intuitive that a function starting with `get` that returns a `shared_ptr` actually returns a *clone* of the manager. I would recommend returning a const reference in the getter, and then using an actual `clone` method or `getCopy` when we need a copy.

How do you feel about fixing the above points? I can draft a PR, but I'm not sure which way you want to go

Levi-Armstrong, 2021-02-03:
@haudren Thank you for the bug report. I have created PR #541 which should address this issue. Do you mind testing it?

Levi-Armstrong, 2021-02-11:
@haudren  I have created a new PR #544 which includes the fix along with cleaning up duplicated code and adds several unit tests with one that confirms this bug is fixed. Let me know if this does not address the issue.

haudren, 2021-02-15:
Issue is fixed, thanks :+1:

closed 2021-02-15 by haudren