tesseract-robotics/tesseract #551: checkJoints does not actually check for joint limits

haudren, 2021-02-19:
While experimenting with OPW Kinematics, I realized that `checkJoints` does not check joint limits. Indeed, even when a joint is detected to be out of limits this `return true` is always hit:
https://github.com/ros-industrial-consortium/tesseract/blob/f33ba348093aa7cc73df6c17450dbad0eeda11c9/tesseract_kinematics/src/opw/opw_inv_kin.cpp#L119

I fixed it in my branch:
https://github.com/haudren/tesseract/commit/cb59289c92ac5f3f152f2ce78ea00cb5f9ed4326

But while adding the tests, I noticed it broke some other tests that are checking that a vector of `numeric_limits<double>::max` returns true when passed to `checkJoints`. So, is it:
- The header description that is wrong ( https://github.com/ros-industrial-consortium/tesseract/blob/f33ba348093aa7cc73df6c17450dbad0eeda11c9/tesseract_kinematics/include/tesseract_kinematics/core/inverse_kinematics.h#L100 )
- The tests that are incorrect

Thanks

Levi-Armstrong, 2021-02-19:
I will go through and fix the `checkJoints` function.

haudren, 2021-03-05:
@Levi-Armstrong : Any progress on this? I can take care of it if you would just let me know which alternative you prefer :)

Levi-Armstrong, 2021-03-05:
Just created a [PR](https://github.com/ros-industrial-consortium/tesseract/pull/564)

haudren, 2021-03-08:
The PR mentioned seems to fix it for OPW, but the newly merged UR kinematics ( #556 ) seems to re-introduce it :thinking:

Levi-Armstrong, 2021-03-08:
Yep, the PR was not updated. PR #565 should fix the issues.

Levi-Armstrong, 2021-03-13:
Please reopen if the issue has not been resolved.

closed 2021-03-13 by Levi-Armstrong