tesseract-robotics/tesseract #559: Threading issues in environment

mpowelson, 2021-03-03:
I made this unit test that fails. It adds and removes a link repeatedly in different threads.
https://github.com/mpowelson/tesseract/tree/bug/threaded_environment

mpowelson, 2021-03-03:
@Levi-Armstrong Do you think you could take a look? Possibly related to https://github.com/ros-industrial-consortium/tesseract_ros/issues/90

mpowelson, 2021-03-03:
[This](https://github.com/mpowelson/tesseract/blob/769c5ad5ee4ee985247bbbcf8d1805d3b1834c36/tesseract_environment/src/core/environment.cpp#L169) seems like a problem. Doesn't this lock immediately go out of scope?

Levi-Armstrong, 2021-03-03:
Fixing it now.

Levi-Armstrong, 2021-03-03:
The last commit in [this](https://github.com/ros-industrial-consortium/tesseract/pull/560) PR should fix the locking issue.

mpowelson, 2021-03-03:
Addressed in #560

closed 2021-03-03 by mpowelson