tesseract-robotics/tesseract #566: What preprocessor macros need to be defined?

ProfInkling, 2021-03-08:
We are using Tesseract on Windows. We build a DLL with Tesseract code, and then link with another DLL with our own code that. We include Tesseract headers into both DLLs. We were getting heap corruption because Tesseract defines the `BT_USE_DOUBLE_PRECISION` macro and we were not defining it for our own code, so some types were different sizes in the different DLLs.

Are there any other macros you know of that we should be defining as well?

Levi-Armstrong, 2021-03-08:
Need a little more information which will help me narrow things down.

- What version of Ubuntu are you on?
- What branch of Tesseract are you using?

ProfInkling, 2021-03-08:
* We are using Windows 10
* We are on commit 4459f6079e9e9ff9de958a19f90eb480739ea18f (master as of January 29)

Levi-Armstrong, 2021-03-09:
Ok, so I assume you are building tesseract_ext to build Bullet from source? If so, I do know that Bullet does not properly export it compiler definitions on past version and maybe master. I recommend checking the BullletConfig.cmake file to see if the compiler definition is listed. If not then I will need to look into Bullet and see how best to create a patch.