tesseract-robotics/tesseract #582: Contact managers not getting updated when adding links

mpowelson, 2021-04-01:
I don't have any shareable code to replicate this, but it seems that when adding links with replace turned on (may not be relevant) at least some times the contact managers don't get correctly updated. I'm not 100% sure this really the case, but something seems off.

Levi-Armstrong, 2021-04-05:
I have found the issue. It is an interesting issue. Bullet does some optimizations to prevent updating things so if you change the active links and not change the contact distance things in the broadphase do not update. In the unit test you created if you change the contact distance to something small like 0.0000000001 it works. Though if you increase this but move set active links after setting the contact distance it no longer works. I did some digging in bullet and found found a solution but need to create a unit test in tesseract collision. My changes are current on top the most recent PR so I create a new PR once it is merged with only the change to tesseract_collision.

mpowelson, 2021-04-05:
That's a scary bug. Thanks for looking into it. Let me know if you need me to do anything.

mpowelson, 2021-04-08:
Fixed in #584

closed 2021-04-08 by mpowelson