tesseract-robotics/tesseract #591: Expand JointLimits

mpowelson, 2021-04-20:
Allow different upper/lower bounds on velocity and acceleration and add jerk limits. This will be useful if we write a wrapper for ruckig, which supports these things.

Levi-Armstrong, 2021-04-21:
Sounds reasonable to me. Do you have a proposed updated to the data structure?

mpowelson, 2021-04-21:
Not really. I was just posting this as a reminder of a potentially breaking change we might want to make soon. However, I would imagine something like this. We could use std::pair, but I kind of like position.lower better than position.first

```
class Bounds
{
   Bounds(double lower, double upper) ;
   Bounds(double upper) : upper(upper) { lower = -upper;}
   double lower;
   double upper;
}

class JointLimits
{
public:

  JointLimits() { this->clear(); }
  Bounds position;
  Bounds effort;
  Bounds velocity;
  Bounds acceleration;
  Bounds jerk;  
...
};

```

Levi-Armstrong, 2024-06-22:
This is now supported

closed 2024-06-22 by Levi-Armstrong