tesseract-robotics/tesseract #603: Tesseract clone performance

haudren, 2021-05-17:
I was recently updating my local copy of Tesseract and noticed a simple fetch took several minutes ( on a decent but far from amazing network connection). After a little bit of prodding, I noticed the following:
- A regular clone from scratch, e.g. `git clone https://github.com/ros-industial-consortium/tesseract` clocks in at almost 1.5GiB and 379062 objects which can take hours to download on some unstable networks.
- A smaller clone, `git clone https://github.com/ros-industial-consortium/tesseract --single-branch` that only retrieves master give me only 27777 objects for a total of 79.99 MiB which seems perfectly reasonable.

I believe the issue is linked to the new (?) `gh_pages` branch that seems to contain a lot of updates and is a pretty big performance drag.

Would it be possible to host the documentation in some other way? I will try to remember to "ignore" that branch in the meantime, but it makes my git flow a bit less agile :sweat_smile:

gavanderhoorn, 2021-05-17:
I've noticed this as well.

I'm surprised GH hasn't complained yet about the size of the repository.

Levi-Armstrong, 2021-05-17:
Thank you for the information. We have plans to move the documentation to its own repository which will happen in the near future. 

What would be a good name for this repository tesseract_docs, tesseract_documentation, etc?

gavanderhoorn, 2021-05-17:
> What would be a good name for this repository tesseract_docs, tesseract_documentation, etc?

both seem OK to me.

`tesseract_docs` is a bit shorter, so might be easier to work with.

haudren, 2021-05-18:
I like `tesseract_docs` as well, it seems to follow a somewhat established practice with the (now defunct) moveit_docs, but also github's own documentation, Laravel's...

Levi-Armstrong, 2022-01-26:
The documentation has been moved but not sure if it resolves the issue.

gavanderhoorn, 2022-01-26:
It's still there in the history. Removing it entirely will require purging that branch from history and (probably) a force-push to this repository.

Levi-Armstrong, 2022-01-26:
I see, so I need to also move the doxygen and benchmark data to the docs repo and then purge the branch. Is that correct?