tesseract-robotics/tesseract #605: IK fast kinematics wrapper only works for Transform6D

mpowelson, 2021-05-18:
ikfast_inv_kin.hpp calls ComputeIK which takes a couple of double*. Depending on what IK type it is, those have different meanings. See [here](https://github.com/rdiankov/openrave/blob/ec02e7ceda3d08e778ad88a0fe298b715a4253bb/python/ikfast.h#L557). It looks like the generated code does have a GetIkType function, but I don't see a key for what those mean. I will say that my generated code for a Ray4D returns `0x46000005`.

Fixing this should just require figuring out what the IK type ints mean and then adding a switch in [ik_fast_kin.hpp](https://github.com/ros-industrial-consortium/tesseract/blob/5b3652689f25c91e6c1796841ab37019cf1f859e/tesseract_kinematics/include/tesseract_kinematics/ikfast/impl/ikfast_inv_kin.hpp#L72) to pass in the angle in the correct format.

zark-zeugan, 2022-06-22:
I wish to help you solve this issue. 
Could you direct me as to how I can get started?

Levi-Armstrong, 2022-06-22:
I will take sometime and post my thoughts on how best to do this since this will also be part of the base interface. This is also related to issue #722.