tesseract-robotics/tesseract #608: Build error related to bullet member functions

MichaelRyanGreer, 2021-06-01:
When I attempt to build tesseract using catkin-tools, the tesseract-collision build fails because it references some functions in "tesseract/tesseract_collision/src/bullet/tesseract_convex_convex_algorithm.cpp" that can't be found. Operating system is Ubuntu 18.05 LTS with ROS Melodic. This is after using wstool and rosdep to handle dependencies. Catkin build output is attached. This is probably a local problem with my build environment, since the build shows as passing on github. Any tips would be appreciated!

[output.txt](https://github.com/ros-industrial-consortium/tesseract/files/6580906/output.txt)

Levi-Armstrong, 2021-06-01:
For 18.04 you will need to clone tesseract_ext repository which will build the correct version of dependencies not available on 18.04. This is not need for 20.04, but unfortunately it is required for 18.04.

MichaelRyanGreer, 2021-06-02:
That fixed it, thanks for the response!

closed 2021-06-02 by MichaelRyanGreer