tesseract-robotics/tesseract #618: Two class names in tesseract_collision are misspelled.

DavidMerzJr, 2021-07-13:
https://github.com/ros-industrial-consortium/tesseract/blob/c2aea9f114719533ade08f9dfe65e0589b8e7933/tesseract_collision/include/tesseract_collision/convex_decomposition/convex_decomposition_vhacd.h#L57

https://github.com/ros-industrial-consortium/tesseract/blob/c2aea9f114719533ade08f9dfe65e0589b8e7933/tesseract_collision/include/tesseract_collision/convex_decomposition/convex_decomposition_hacd.h#L47

The file names and the base class all use the expected spelling, Convex De**c**ompsition.  However, the actual class names for both the HACD and VHACD convex decomposition implementations are spelled without the 'c', as Convex De omposition.

ConvexDe**c**ompositionVHACD
ConvexDeompositionVHACD

ConvexDe**c**ompositionHACD
ConvexDeompositionHACD

@Levi-Armstrong Is there a reason for this, or should I open a pull request?

DavidMerzJr, 2021-07-13:
This was resolved by #620

closed 2021-07-13 by DavidMerzJr

Levi-Armstrong, 2021-07-13:
@DavidMerzJr Thanks for the fix!