tesseract-robotics/tesseract #621: Build error in tesseract_collision_convex_decomposition

johnwason, 2021-07-14:
I am getting this error building on Ubuntu 20.04 ROS Noetic gcc 9.3.0:

```
Errors     << tesseract_collision:make /home/user/catkin_ws_tesseract2/logs/tesseract_collision/build.make.000.log                                                            
In file included from /home/user/catkin_ws_tesseract2/src/tesseract/tesseract_collision/src/convex_decomposition/convex_decomposition_hacd.cpp:4:
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdCircularList.h: In instantiation of ‘HACD::CircularListElement<T>::CircularListElement(const T&) [with T = HACD::TMMVertex]’:
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdCircularList.inl:75:14:   required from ‘HACD::CircularListElement<T>* HACD::CircularList<T>::Add(const T*) [with T = HACD::TMMVertex]’
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdManifoldMesh.h:177:70:   required from here
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdCircularList.h:34:46: error: implicitly-declared ‘HACD::TMMVertex& HACD::TMMVertex::operator=(const HACD::TMMVertex&)’ is deprecated [-Werror=deprecated-copy]
   34 |  CircularListElement(const T& data) { m_data = data; }
      |                                       ~~~~~~~^~~~~~
In file included from /home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdICHull.h:19,
                 from /home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdGraph.h:20,
                 from /home/user/catkin_ws_tesseract2/src/tesseract/tesseract_collision/src/convex_decomposition/convex_decomposition_hacd.cpp:5:
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdManifoldMesh.h:83:2: note: because ‘HACD::TMMVertex’ has user-provided ‘HACD::TMMVertex::TMMVertex(const HACD::TMMVertex&)’
   83 |  TMMVertex(const TMMVertex &rhs);
      |  ^~~~~~~~~
In file included from /home/user/catkin_ws_tesseract2/src/tesseract/tesseract_collision/src/convex_decomposition/convex_decomposition_hacd.cpp:4:
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdCircularList.h: In instantiation of ‘HACD::CircularListElement<T>::CircularListElement(const T&) [with T = HACD::TMMEdge]’:
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdCircularList.inl:75:14:   required from ‘HACD::CircularListElement<T>* HACD::CircularList<T>::Add(const T*) [with T = HACD::TMMEdge]’
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdManifoldMesh.h:179:63:   required from here
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdCircularList.h:34:46: error: implicitly-declared ‘constexpr HACD::TMMEdge& HACD::TMMEdge::operator=(const HACD::TMMEdge&)’ is deprecated [-Werror=deprecated-copy]
   34 |  CircularListElement(const T& data) { m_data = data; }
      |                                       ~~~~~~~^~~~~~
In file included from /home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdICHull.h:19,
                 from /home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdGraph.h:20,
                 from /home/user/catkin_ws_tesseract2/src/tesseract/tesseract_collision/src/convex_decomposition/convex_decomposition_hacd.cpp:5:
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdManifoldMesh.h:105:2: note: because ‘HACD::TMMEdge’ has user-provided ‘HACD::TMMEdge::TMMEdge(const HACD::TMMEdge&)’
  105 |  TMMEdge(const TMMEdge &rhs);
      |  ^~~~~~~
In file included from /home/user/catkin_ws_tesseract2/src/tesseract/tesseract_collision/src/convex_decomposition/convex_decomposition_hacd.cpp:4:
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdCircularList.h: In instantiation of ‘HACD::CircularListElement<T>::CircularListElement(const T&) [with T = HACD::TMMTriangle]’:
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdCircularList.inl:75:14:   required from ‘HACD::CircularListElement<T>* HACD::CircularList<T>::Add(const T*) [with T = HACD::TMMTriangle]’
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdManifoldMesh.h:181:75:   required from here
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdCircularList.h:34:46: error: implicitly-declared ‘HACD::TMMTriangle& HACD::TMMTriangle::operator=(const HACD::TMMTriangle&)’ is deprecated [-Werror=deprecated-copy]
   34 |  CircularListElement(const T& data) { m_data = data; }
      |                                       ~~~~~~~^~~~~~
In file included from /home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdICHull.h:19,
                 from /home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdGraph.h:20,
                 from /home/user/catkin_ws_tesseract2/src/tesseract/tesseract_collision/src/convex_decomposition/convex_decomposition_hacd.cpp:5:
/home/user/catkin_ws_tesseract2_ext/devel/include/bullet/HACD/hacdManifoldMesh.h:128:2: note: because ‘HACD::TMMTriangle’ has user-provided ‘HACD::TMMTriangle::TMMTriangle(const HACD::TMMTriangle&)’
  128 |  TMMTriangle(const TMMTriangle &rhs);
      |  ^~~~~~~~~~~
cc1plus: some warnings being treated as errors
make[2]: *** [CMakeFiles/tesseract_collision_convex_decomposition.dir/build.make:76: CMakeFiles/tesseract_collision_convex_decomposition.dir/src/convex_decomposition/convex_decomposition_hacd.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:160: CMakeFiles/tesseract_collision_convex_decomposition.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
```

Levi-Armstrong, 2021-07-14:
What version of Bullet are you using?

johnwason, 2021-07-14:
I am using bullet3 from tesseract_ext

Levi-Armstrong, 2021-07-14:
So tesseract_ext is currently pointing to the master branch which must have deprecated some function. It should be updated to point to the version that is currently released on Ubuntu 20.04.

johnwason, 2021-07-14:
The package search is showing bullet version 2.88

```
Package: libbullet-dev
Version: 2.88+dfsg-2build2
Priority: extra
Section: universe/libdevel
Source: bullet
Origin: Ubuntu
```

johnwason, 2021-07-14:
This is the file causing the problem. It hasn't been modified since 2018

https://github.com/bulletphysics/bullet3/blob/master/Extras/HACD/hacdCircularList.h

johnwason, 2021-07-16:
I removed tesseract_ext and installed the packages as specified in the readme. I had to also install `ros-noetic-taskflow` which is not listed in the readme. GCC will ignore errors in warnings in "system directories". When bullet is installed on the system, the warnings as errors are ignored. When I built in tesseract_ext, bullet was not in a "system directory", so the errors showed in the build. CMake and GCC can be configured  to treat additional directories as system directories, but it doesn't appear that the normal `find_package` modules do this.

closed 2021-07-16 by johnwason

Levi-Armstrong, 2021-07-17:
I will try building from source and see how to resolve the issue. All of the CI dockers have it installed from source which explains why they are working.