tesseract-robotics/tesseract #626: Tesseract sometimes stops detecting any collisions

joelwilliamsonoctopuz, 2021-08-10:
We are using Tesseract for collision detection (and kinematics) in our software. Recently we noticed a bug in which certain robot poses where the robot was in collision with a component in our scene would cause Tesseract to stop reporting any collision errors for any robot pose. When this happens, the only way to get it working again is to create a new BulletDiscreteBVHManager.

I can provide the geometries we saw trigger this bug and the transformation matrices for each geometry, or an example program that reproduces the issue. Would either be helpful for you to debug this?

We are building this on Windows which could be related, but hopefully this is reproducible on Linux.

Levi-Armstrong, 2021-08-10:
An example program which reproduce the issue would allow me to track down the issue faster. Then I could also add it as a unit test.

Levi-Armstrong, 2021-08-21:
@joelwilliamsonoctopuz I wanted to follow up to see if you resolved the issue?

joelwilliamsonoctopuz, 2021-08-23:
@Levi-Armstrong Sorry about the delay. Our product is a mix of C# and C++, so the simplest reproduction we have wouldn't work for you. I tried to reproduce the error in pure C++ so you could run it, but I wasn't able to. Would you mind having a video chat with me so I can show you what I'm doing?

Levi-Armstrong, 2021-08-23:
Sure. I am free today between 1-2pm CT and after 3pm CT.

joelwilliamsonoctopuz, 2021-08-23:
1pm CT works for me. What video chat program do you use?

joelwilliamsonoctopuz, 2021-08-24:
[ply_files_for_Levi.zip](https://github.com/ros-industrial-consortium/tesseract/files/7041343/ply_files_for_Levi.zip)

Hey Levi, yesterday was really helpful. Thank you very much. Here are the ply files for the parts I was using. The shape containing all the "collision_object_2_*.ply" meshes use an identity transform for all the meshes. The transforms for the parts in collision are:
                ```collision_object_1: new Matrix4x4(new Vec4(-0.70581769378301, -0.686451318220594, -0.174945842562787, 0), new Vec4(0.654932510475525, -0.726456551332563, 0.208144939531274, 0), new Vec4(-0.269971912875806, 0.0323346590066556, 0.962325128488445, 0), new Vec4(970.507782716257, -199.022472197488, 5.7890632254045, 1))
                collision_object_2: new Matrix4x4(new Vec4(1, 0, 0, 0), new Vec4(0, 1, 0, 0), new Vec4(0, 0, 1, 0), new Vec4(1750, 250, -20, 1))```

and the transforms for the parts out of collision are: 
                ```collision_object_1: Matrix4x4(new Vec4(-0.706382516470308, 0.0148600750052246, -0.707674299818655, 0), new Vec4(-0.0510841828014138, -0.998242777458575, 0.0300293666268319, 0), new Vec4(-0.70598454856475, 0.057363185449714, 0.705900363136415, 0), new Vec4(979.239218615628, -85.6723302587343, 518.894463458944, 1))
                collision_object_2: Matrix4x4(new Vec4(1, 0, 0, 0), new Vec4(0, 1, 0, 0), new Vec4(0, 0, 1, 0), new Vec4(1750, 250, -20, 1))```

joelwilliamsonoctopuz, 2021-08-31:
@Levi-Armstrong have you had any luck reproducing this on yourself? We've switch to FCL and it is working well.

Levi-Armstrong, 2021-12-15:
@joelwilliamsonoctopuz Has this been resolved?

Levi-Armstrong, 2022-01-26:
Please reopen if not resolved.

closed 2022-01-26 by Levi-Armstrong