tesseract-robotics/tesseract #673: undefined reference to the function of Bullet when building tesseract_collision

xibeisiber, 2021-12-02:
Hi,
When I was building tesseract_collision I encountered the following error:
`Errors     << tesseract_collision:make /home/jia/trajplan_ws/logs/tesseract_collision/build.make.009.log                                   
libtesseract_collision_bullet.so: undefined reference to `btCollisionShape::getBoundingSphere(btVector3&, double&) const'
libtesseract_collision_bullet.so: undefined reference to `btConvexShape::project(btTransform const&, btVector3 const&, double&, double&, btVector3&, btVector3&) const'
libtesseract_collision_bullet.so: undefined reference to `btPolyhedralConvexShape::setPolyhedralFeatures(btConvexPolyhedron&)'
libtesseract_collision_bullet.so: undefined reference to `btCollisionShape::getContactBreakingThreshold(double) const'
libtesseract_collision_bullet.so: undefined reference to `btPolyhedralContactClipping::clipFaceAgainstHull(btVector3 const&, btConvexPolyhedron const&, btTransform const&, btAlignedObjectArray<btVector3>&, btAlignedObjectArray<btVector3>&, double, double, btDiscreteCollisionDetectorInterface::Result&)'
libtesseract_collision_bullet.so: undefined reference to `btPolyhedralContactClipping::clipHullAgainstHull(btVector3 const&, btConvexPolyhedron const&, btConvexPolyhedron const&, btTransform const&, btTransform const&, double, double, btAlignedObjectArray<btVector3>&, btAlignedObjectArray<btVector3>&, btDiscreteCollisionDetectorInterface::Result&)'
libtesseract_collision_bullet.so: undefined reference to `gDbvtMargin'
libtesseract_collision_bullet.so: undefined reference to `btConvexPolyhedron::initialize2()'
libtesseract_collision_bullet.so: undefined reference to `btConeShapeZ::btConeShapeZ(double, double)'
libtesseract_collision_bullet.so: undefined reference to `btCapsuleShapeZ::btCapsuleShapeZ(double, double)'
libtesseract_collision_bullet.so: undefined reference to `btConvexHullComputer::compute(void const*, bool, int, int, double, double)'
libtesseract_collision_bullet.so: undefined reference to `btConvexHullShape::btConvexHullShape(double const*, int, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [bullet/create_convex_hull] Error 1
make[1]: *** [bullet/CMakeFiles/create_convex_hull.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
../bullet/libtesseract_collision_bullet.so: undefined reference to `btCollisionShape::getBoundingSphere(btVector3&, double&) const'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btConvexShape::project(btTransform const&, btVector3 const&, double&, double&, btVector3&, btVector3&) const'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btPolyhedralConvexShape::setPolyhedralFeatures(btConvexPolyhedron&)'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btCollisionShape::getContactBreakingThreshold(double) const'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btPolyhedralContactClipping::clipFaceAgainstHull(btVector3 const&, btConvexPolyhedron const&, btTransform const&, btAlignedObjectArray<btVector3>&, btAlignedObjectArray<btVector3>&, double, double, btDiscreteCollisionDetectorInterface::Result&)'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btPolyhedralContactClipping::clipHullAgainstHull(btVector3 const&, btConvexPolyhedron const&, btConvexPolyhedron const&, btTransform const&, btTransform const&, double, double, btAlignedObjectArray<btVector3>&, btAlignedObjectArray<btVector3>&, btDiscreteCollisionDetectorInterface::Result&)'
../bullet/libtesseract_collision_bullet.so: undefined reference to `gDbvtMargin'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btConvexPolyhedron::initialize2()'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btConeShapeZ::btConeShapeZ(double, double)'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btCapsuleShapeZ::btCapsuleShapeZ(double, double)'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btConvexHullComputer::compute(void const*, bool, int, int, double, double)'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btConvexHullShape::btConvexHullShape(double const*, int, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/tesseract_collision_box_box_example] Error 1
make[1]: *** [examples/CMakeFiles/tesseract_collision_box_box_example.dir/all] Error 2
make: *** [all] Error 2
cd /home/jia/trajplan_ws/build/tesseract_collision; catkin build --get-env tesseract_collision | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................
Failed     << tesseract_collision:make                        [ Exited with code 2 ]                                                       
Failed    <<< tesseract_collision`

The output when `ldd libtesseract_collision_bullet.so` is:
`	linux-vdso.so.1 (0x00007fff4b3e9000)
	libtesseract_collision_core.so => /home/jia/trajplan_ws/build/tesseract_collision/core/libtesseract_collision_core.so (0x00007f9295f09000)
	libLinearMath.so.2.87 => /usr/lib/x86_64-linux-gnu/libLinearMath.so.2.87 (0x00007f9295ce7000)
	libBulletCollision.so.2.87 => /usr/lib/x86_64-linux-gnu/libBulletCollision.so.2.87 (0x00007f92959e8000)
	libtesseract_common.so => /home/jia/trajplan_ws/devel/lib/libtesseract_common.so (0x00007f9295523000)
	libconsole_bridge.so.0.4 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007f929531e000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007f9295119000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f9294d90000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f92949f2000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f92947da000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f92943e9000)
	libyaml-cpp.so.0.5 => /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.5 (0x00007f929416f000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f9293f6b000)
	libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007f9293d51000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f929665b000)
	libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007f9293b3d000)
	libboost_serialization.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1 (0x00007f92938ff000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f92936e0000)`

I have Bullet 2.87 installed when installing ros and gazebo, and downloaded tesserect_ext in which bullet 3.21 is installed in the ros workspace "/home/jia/trajplan_ws/devel". 
Should tesseract_collision use bullet 3.21? If so, how to set the version in the CMakeList.txt of tesseract_collision ?

Levi-Armstrong, 2021-12-02:
What version of Ubuntu are you using?

xibeisiber, 2021-12-02:
> What version of Ubuntu are you using?

ubuntu 18.04. ros melodic

xibeisiber, 2021-12-02:
Maybe the reason is that  USE_DOUBLE_PRECISION is off in the system-installed Bullet 2.87?

Levi-Armstrong, 2021-12-02:
assuming from the version it looks like you maybe on 18.04. If this is the case then you need to include the [tesseract_ext](https://github.com/tesseract-robotics/tesseract_ext) repository which will pull in newer versions of bullet, fcl, and other dependencies which are not available. If you are on Ubuntu 20.04 this this repository is no longer needed.

xibeisiber, 2021-12-02:
> 

I have built the tesseract_ext, and bullet 3.21 is built within my ros workspace. But tesseract_collision failed to find the bullet3.21 when building.

Levi-Armstrong, 2021-12-02:
So tesseract_ext should build version 2.89 with double precision enabled and not 3.21. Was something modified?

xibeisiber, 2021-12-03:
I built bullet 3.21 with double precision enabled, installed it in /usr/local, and add "3.21" in the "find_package" of  tesseract_collision/bullet/CMakeLists.txt..
Now tesseract_collision can be builted successfully.

closed 2021-12-08 by xibeisiber