tesseract-robotics/tesseract #677: error at "env_->init(urdf_xml_string, srdf_xml_string, locator)"

xibeisiber, 2021-12-08:
Hi, I am trying to implement a planning node for iiwa following the codes in tesseract_ros_example.
But the node dies at `env_->init(urdf_xml_string, srdf_xml_string, locator)`. The error info is:
`terminate called after throwing an instance of 'std::runtime_error'
  what():  ContactManagersPluginFactory, tried to get default discrete contact manager but none exist!`.

Does the content in "/robot_description" or "/robot_description_semantic" (see attached) cause this?
I can provide more information if needed.
[robot_description.txt](https://github.com/tesseract-robotics/tesseract/files/7672615/robot_description.txt)

[robot_description_semantic.txt](https://github.com/tesseract-robotics/tesseract/files/7672616/robot_description_semantic.txt)

xibeisiber, 2021-12-08:
found the reason. I was using the srdf file generated by moveit_setup_assistant, so this file does not include the `contact_managers_plugin_config`...

closed 2021-12-08 by xibeisiber

Levi-Armstrong, 2021-12-08:
That is correct. Also there is one for the kinematics if you need inverse kinematics.

xibeisiber, 2021-12-09:
> That is correct. Also there is one for the kinematics if you need inverse kinematics.

ok. Thanks!