tesseract-robotics/tesseract #686: tesseract_common `FindTinyXML2.cmake` interferes with the `tinyxml2_vendor` ROS2 package

jdlangs, 2021-12-15:
It appear if tesseract_common is found in a CMake environment first, it interferes with some variables that the vendor package uses, which defines its own `FindTinyXML2.cmake`. You get an error like:

```
CMake Error at /opt/ros/galactic/share/tinyxml2_vendor/cmake/tinyxml2_vendor-extras.cmake:23 (message):
  Unable to extract the library file path from
  /usr/lib/x86_64-linux-gnu/libtinyxml2.so
Call Stack (most recent call first):
  /opt/ros/galactic/share/tinyxml2_vendor/cmake/tinyxml2_vendorConfig.cmake:41 (include)
  /opt/ros/galactic/share/pluginlib/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/galactic/share/pluginlib/cmake/pluginlibConfig.cmake:41 (include)
  CMakeLists.txt:19 (find_package)
```

because the `TINYXML2_LIBRARY` variable uses a different convention. The reverse seems to be ok, if the vendor package goes first. Any thoughts on the best approach to resolve this conflict?

Levi-Armstrong, 2021-12-16:
I am not understanding what the issue is. Could you provide more details?

jdlangs, 2021-12-16:
The issue is if you `find_package(tesseract_common)` and then `find_package(pluginlib)` (or another ROS package with uses `tinyxml2_vendor`) you get an error. I believe the root issue is both packages provide a `FindTinyXML2.cmake` file ([here](https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_common/cmake/FindTinyXML2.cmake) and [here](https://github.com/ros2/tinyxml2_vendor/blob/master/cmake/Modules/FindTinyXML2.cmake)) which are different. And if the tesseract one goes first, something breaks in the `tinyxml2_vendor` CMake logic.

I'm not sure what options there are to avoid this conflict. Can some sort of scope be introduced so each package can find `TinyXML2` however it wants to?

Levi-Armstrong, 2021-12-16:
Is the error because there are two variables with the same name?

Levi-Armstrong, 2022-05-31:
@jdlangs I believe there find tinyxml2 is better and if we replaced ours with it we should not run into this issue.

Levi-Armstrong, 2022-05-31:
This should be resolved by PR #774