tesseract-robotics/tesseract #699: Ubuntu 18 ROS Melodic build error: /usr/bin/ld: cannot find -lBulletInverseDynamics

Davidlequnchen, 2022-01-17:
I'm trying to build the package with tesseract_ext, tesseract_gui, tesseract_ros, and tesseract_planning.
The platform is Ubuntu 18.04 and ROS Melodic. After following the instructions from the documentation and "catkin build", I got the following error:
Errors     << tesseract_collision:make /home/chenlequn/tesseract_ws/logs/tesseract_collision/build.make.002.log                                                             
/usr/bin/ld: cannot find -lBulletInverseDynamics
collect2: error: ld returned 1 exit status
make[2]: *** [bullet/libtesseract_collision_bullet.so] Error 1
make[1]: *** [bullet/CMakeFiles/tesseract_collision_bullet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/chenlequn/tesseract_ws/build/tesseract_collision; catkin build --get-env tesseract_collision | catkin env -si  /usr/bin/make --jobserver-fds=4,5 -j; cd -

............................................................................................................................................................................
Failed     << tesseract_collision:make               [ Exited with code 2 ]                                                                                                 
Failed    <<< tesseract_collision                    [ 5.1 seconds ]                                                                                                        
Abandoned <<< tesseract_urdf                         [ Unrelated job failed ]                                                                                               
Abandoned <<< tesseract_state_solver                 [ Unrelated job failed ]                                                                                               
Abandoned <<< tesseract_kinematics                   [ Unrelated job failed ]                                                                                               
Abandoned <<< tesseract_environment                  [ Unrelated job failed ]                                                                                               
Abandoned <<< tesseract_visualization                [ Unrelated job failed ]   

[image]

Levi-Armstrong, 2022-01-17:
On 20.04 all of the dependencies are available but on 18.04 they are not. As a result you have to clone the [tesseract_ext](https://github.com/tesseract-robotics/tesseract_ext) repository into your workspace which will pull the right dependencies like bullet and fcl for the correct version and build them in your workspace.

Davidlequnchen, 2022-01-18:
> 

Thank you for your response. After I add tesseract_ext package in my workspace, I still got the same error:
[image]

And the packages that I cloned to my workspace are:
[image]
Any ideas on where this error comes from?

Levi-Armstrong, 2022-01-18:
I would try running 'catkin clean' then build again.

Davidlequnchen, 2022-01-18:
> 

Yes, I have tried that. But unfortunately, I got new errors this time.

 ../bullet/libtesseract_collision_bullet.so: undefined reference to `btConvexHullShape::btConvexHullShape(double const*, int, int)'

I saw someone also got this error in previous issue. I have cloned the tesseract_ext already.

Levi-Armstrong, 2022-01-18:
Could you provide your whole build log from the  terminal after a clean build?

Davidlequnchen, 2022-01-18:
> Could you provide your whole build log from the terminal after a clean build?
catkin build 
-----------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/chenlequn/tesseract_ws
-----------------------------------------------------------------
Build Space:        [exists] /home/chenlequn/tesseract_ws/build
Devel Space:        [exists] /home/chenlequn/tesseract_ws/devel
Install Space:      [unused] /home/chenlequn/tesseract_ws/install
Log Space:          [exists] /home/chenlequn/tesseract_ws/logs
Source Space:       [exists] /home/chenlequn/tesseract_ws/src
DESTDIR:            [unused] None
-----------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-----------------------------------------------------------------
[build] Found '39' packages in 0.0 seconds.                                    
[build] Package table is up to date.                                           
Starting  >>> bullet                                                           
Starting  >>> eigen                                                            
Starting  >>> libccd                                                           
Starting  >>> osqp                                                             
Starting  >>> swig                                                             
Starting  >>> taskflow                                                         
Starting  >>> tesseract_msgs                                                   
Finished  <<< eigen                                          [ 0.1 seconds ]   
Starting  >>> descartes_light                                                  
Starting  >>> ifopt                                                            
Starting  >>> ompl                                                             
Starting  >>> orocos_kdl                                                       
Starting  >>> tesseract_common                                                 
Starting  >>> trajopt_utils                                                    
_______________________________________________________________________________
Warnings   << swig:cmake /home/chenlequn/tesseract_ws/logs/swig/build.cmake.001.log
Swig version found: 3.0.12
cd /home/chenlequn/tesseract_ws/build/swig; catkin build --get-env swig | catkin env -si  /usr/bin/cmake /home/chenlequn/tesseract_ws/src/tesseract_ext-master/swig --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/home/chenlequn/tesseract_ws/devel; cd -

...............................................................................
Finished  <<< taskflow                                       [ 0.1 seconds ]   
Finished  <<< libccd                                         [ 0.2 seconds ]   
Starting  >>> fcl                                                              
Finished  <<< osqp                                           [ 0.1 seconds ]   
Starting  >>> osqp_eigen                                                       
Finished  <<< ifopt                                          [ 0.4 seconds ]   
Finished  <<< descartes_light                                [ 0.4 seconds ]   
Finished  <<< ompl                                           [ 0.2 seconds ]   
_______________________________________________________________________________
Warnings   << tesseract_msgs:cmake /home/chenlequn/tesseract_ws/logs/tesseract_msgs/build.cmake.001.log
CMake Warning (dev) at /usr/src/googletest/CMakeLists.txt:3 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
[... 172 lines cut ...]
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
cd /home/chenlequn/tesseract_ws/build/swig; catkin build --get-env swig | catkin env -si  /usr/bin/make --jobserver-fds=4,5 -j; cd -

............................................................................................................................................................................
____________________________________________________________________________________________________________________________________________________________________________
Warnings   << swig:install /home/chenlequn/tesseract_ws/logs/swig/build.install.001.log                                                                                     
+ test -d Tools/config
+ aclocal -I Tools/config
+ autoheader
+ automake --add-missing --copy --force-missing
+ autoconf
+ cd CCache
+ autoreconf
cd /home/chenlequn/tesseract_ws/build/swig; catkin build --get-env swig | catkin env -si  /usr/bin/make install; cd -

[... cut ...]
[build]   Warnings:  7 packages succeeded with warnings.                                                                                                                    
[build]   Abandoned: 15 packages were abandoned.                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                     
[build] Runtime: 19.3 seconds total.                                 

[image]

Levi-Armstrong, 2022-01-18:
It looks like for some reason it did not build bullet because it only took 5.5s. I am not sure why because it should have checked and determined it was needed. You can force it to build by passing the following `--cmake-args -DINSTALL_BULLET=ON` and may need to do the same for fcl because I did not see it the log above so not sure if it was built.

Davidlequnchen, 2022-01-19:
> Just in case I would go ahead and force it to also build FCL `--cmake-args -DINSTALL_BULLET=ON -DINSTALL_FCL=ON`

Still got the same error after I have tried:
`catkin clean`
`catkin build --cmake-args -DINSTALL_BULLET=ON -DINSTALL_FCL=ON`

Below is the build log:
```text
Finished  <<< swig                                           [ 1 minute and 49.3 seconds ]                                                                                  
Finished  <<< tesseract_scene_graph                          [ 33.4 seconds ]                                                                                               
Starting  >>> tesseract_srdf                                                                                                                                                
Finished  <<< tesseract_srdf                                 [ 24.6 seconds ]                                                                                               
____________________________________________________________________________________________________________________________________________________________________________
Warnings   << fcl:make /home/chenlequn/tesseract_ros_ws/logs/fcl/build.make.000.log                                                                                         
Cloning into 'fcl-src'...
Note: checking out '0.6.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 97455a4 FCL 0.6.1
cd /home/chenlequn/tesseract_ros_ws/build/fcl; catkin build --get-env fcl | catkin env -si  /usr/bin/make --jobserver-fds=4,5 -j; cd -

............................................................................................................................................................................
Finished  <<< fcl                                            [ 2 minutes and 27.3 seconds ]                                                                                 
____________________________________________________________________________________________________________________________________________________________________________
Warnings   << bullet:make /home/chenlequn/tesseract_ros_ws/logs/bullet/build.make.000.log                                                                                   
Cloning into 'bullet3-src'...
Checking out files:  16% (1272/7537)   
Checking out files:  17% (1282/7537)   
Checking out files:  18% (1357/7537)   
Checking out files:  19% (1433/7537)   
Checking out files:  20% (1508/7537)   
Checking out files:  21% (1583/7537)   
Checking out files:  22% (1659/7537)   
Checking out files:  23% (1734/7537)   
Checking out files:  24% (1809/7537)   
Checking out files:  25% (1885/7537)   
Checking out files:  26% (1960/7537)   
Checking out files:  27% (2035/7537)   
Checking out files:  28% (2111/7537)   
Checking out files:  29% (2186/7537)   
Checking out files:  30% (2262/7537)   
Checking out files:  31% (2337/7537)   
Checking out files:  32% (2412/7537)   
Checking out files:  33% (2488/7537)   
Checking out files:  34% (2563/7537)   
Checking out files:  35% (2638/7537)   
Checking out files:  36% (2714/7537)   
Checking out files:  37% (2789/7537)   
Checking out files:  38% (2865/7537)   
Checking out files:  38% (2898/7537)   
Checking out files:  39% (2940/7537)   
Checking out files:  40% (3015/7537)   
Checking out files:  41% (3091/7537)   
Checking out files:  42% (3166/7537)   
Checking out files:  43% (3241/7537)   
Checking out files:  44% (3317/7537)   
Checking out files:  45% (3392/7537)   
Checking out files:  46% (3468/7537)   
Checking out files:  47% (3543/7537)   
Checking out files:  48% (3618/7537)   
Checking out files:  49% (3694/7537)   
Checking out files:  50% (3769/7537)   
[... 70 lines cut ...]
____________________________________________________________________________________________________________________________________________________________________________
Warnings   << vhacd:cmake /home/chenlequn/tesseract_ros_ws/logs/vhacd/build.cmake.000.log                                                                                   
NO_OPENCL OFF
cd /home/chenlequn/tesseract_ros_ws/build/vhacd; catkin build --get-env vhacd | catkin env -si  /usr/bin/cmake /home/chenlequn/tesseract_ros_ws/src/trajopt/trajopt_ext/vhacd --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/home/chenlequn/tesseract_ros_ws/devel -DINSTALL_BULLET=ON -DINSTALL_FCL=ON -DINSTALL_VHACD=ON; cd -

............................................................................................................................................................................
____________________________________________________________________________________________________________________________________________________________________________
Warnings   << tesseract_collision:cmake /home/chenlequn/tesseract_ros_ws/logs/tesseract_collision/build.cmake.000.log                                                       
Building FCL components
Building VHACD components
NO_OPENCL OFF
[... cut ...]
libtesseract_collision_bullet.so: undefined reference to `btPolyhedralContactClipping::clipHullAgainstHull(btVector3 const&, btConvexPolyhedron const&, btConvexPolyhedron const&, btTransform const&, btTransform const&, double, double, btAlignedObjectArray<btVector3>&, btAlignedObjectArray<btVector3>&, btDiscreteCollisionDetectorInterface::Result&)'
```
I found there might be something wrong with vhacd, so I build again by `catkin build --cmake-args -DINSTALL_BULLET=ON -DINSTALL_FCL=ON -DINSTALL_VHACD=ON`, but still got the same error as above

Davidlequnchen, 2022-01-19:
I forget to mentioned that in __ifopt/ifopt_ipopt__ pakcage, I commented out `find_package(IPOPT 3.11.9 REQUIRED)`, and changed it to `find_package(IPOPT REQUIRED)`.
This is because  `find_package(IPOPT 3.11.9 REQUIRED)` throw me errors like something below:
[image]

Levi-Armstrong, 2022-01-19:
I am going to have to pull down a docker and see what the issue is. The CI uses a docker that I have manually install the dependencies so they are not built every time so it is possible something is not correct with repo.

Davidlequnchen, 2022-01-20:
> 

Thanks for your suggestions. But I still got the same error. 

```text
Errors     << tesseract_collision:make /home/chenlequn/tesseract_ros_ws/logs/tesseract_collision/build.make.000.log
libtesseract_collision_bullet.so: undefined reference to `btCollisionShape::getBoundingSphere(btVector3&, double&) const'
libtesseract_collision_bullet.so: undefined reference to `btConvexShape::project(btTransform const&, btVector3 const&, double&, double&, btVector3&, btVector3&) const'
libtesseract_collision_bullet.so: undefined reference to `btPolyhedralConvexShape::setPolyhedralFeatures(btConvexPolyhedron&)'
libtesseract_collision_bullet.so: undefined reference to `btCollisionShape::getContactBreakingThreshold(double) const'
libtesseract_collision_bullet.so: undefined reference to `btPolyhedralContactClipping::clipFaceAgainstHull(btVector3 const&, btConvexPolyhedron const&, btTransform const&, btAlignedObjectArray<btVector3>&, btAlignedObjectArray<btVector3>&, double, double, btDiscreteCollisionDetectorInterface::Result&)'
libtesseract_collision_bullet.so: undefined reference to `btPolyhedralContactClipping::clipHullAgainstHull(btVector3 const&, btConvexPolyhedron const&, btConvexPolyhedron const&, btTransform const&, btTransform const&, double, double, btAlignedObjectArray<btVector3>&, btAlignedObjectArray<btVector3>&, btDiscreteCollisionDetectorInterface::Result&)'
libtesseract_collision_bullet.so: undefined reference to `gDbvtMargin'
libtesseract_collision_bullet.so: undefined reference to `btConvexPolyhedron::initialize2()'
libtesseract_collision_bullet.so: undefined reference to `btConeShapeZ::btConeShapeZ(double, double)'
libtesseract_collision_bullet.so: undefined reference to `btCapsuleShapeZ::btCapsuleShapeZ(double, double)'
libtesseract_collision_bullet.so: undefined reference to `btConvexHullComputer::compute(void const*, bool, int, int, double, double)'
libtesseract_collision_bullet.so: undefined reference to `btConvexHullShape::btConvexHullShape(double const*, int, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [bullet/create_convex_hull] Error 1
make[1]: *** [bullet/CMakeFiles/create_convex_hull.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
../bullet/libtesseract_collision_bullet.so: undefined reference to `btCollisionShape::getBoundingSphere(btVector3&, double&) const'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btConvexShape::project(btTransform const&, btVector3 const&, double&, double&, btVector3&, btVector3&) const'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btPolyhedralConvexShape::setPolyhedralFeatures(btConvexPolyhedron&)'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btCollisionShape::getContactBreakingThreshold(double) const'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btPolyhedralContactClipping::clipFaceAgainstHull(btVector3 const&, btConvexPolyhedron const&, btTransform const&, btAlignedObjectArray<btVector3>&, btAlignedObjectArray<btVector3>&, double, double, btDiscreteCollisionDetectorInterface::Result&)'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btPolyhedralContactClipping::clipHullAgainstHull(btVector3 const&, btConvexPolyhedron const&, btConvexPolyhedron const&, btTransform const&, btTransform const&, double, double, btAlignedObjectArray<btVector3>&, btAlignedObjectArray<btVector3>&, btDiscreteCollisionDetectorInterface::Result&)'
../bullet/libtesseract_collision_bullet.so: undefined reference to `gDbvtMargin'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btConvexPolyhedron::initialize2()'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btConeShapeZ::btConeShapeZ(double, double)'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btCapsuleShapeZ::btCapsuleShapeZ(double, double)'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btConvexHullComputer::compute(void const*, bool, int, int, double, double)'
../bullet/libtesseract_collision_bullet.so: undefined reference to `btConvexHullShape::btConvexHullShape(double const*, int, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/tesseract_collision_box_box_example] Error 1
make[1]: *** [examples/CMakeFiles/tesseract_collision_box_box_example.dir/all] Error 2
make: *** [all] Error 2
cd /home/chenlequn/tesseract_ros_ws/build/tesseract_collision; catkin build --get-env tesseract_collision | catkin env -si  /usr/bin/make --jobserver-fds=4,5 -j; cd -
```

Seems like the error comes from `make[2]: *** [examples/tesseract_collision_box_box_example] Error 1
make[1]: *** [examples/CMakeFiles/tesseract_collision_box_box_example.dir/all] Error 2
`

Levi-Armstrong, 2022-01-21:
I was able to build on melodic but need to make some changes. It works if you manually install bullet but building with tesseract_ext no longer works because the following needs to be added `target_link_directories(${PROJECT_NAME}_bullet "${BULLET_ROOT_DIR}/${BULLET_LIBRARY_DIRS}")` and `target_link_directories(${PROJECT_NAME}_vhacd "${BULLET_ROOT_DIR}/${BULLET_LIBRARY_DIRS}")`. Also you will need to install a newer version of cmake `pip install cmake` for it to build.

Davidlequnchen, 2022-01-21:
> 

Sorry I'm a bit confused. Where should I add `target_link_directories(${PROJECT_NAME}_bullet "${BULLET_ROOT_DIR}/${BULLET_LIBRARY_DIRS}")` and `target_link_directories(${PROJECT_NAME}_vhacd "${BULLET_ROOT_DIR}/${BULLET_LIBRARY_DIRS}")`?

I just manually installed bullet from `https://github.com/bulletphysics/bullet3`, then compile it using cmake, with `-DUSE_DOUBLE_PRECISION:BOOL=ON -DBULLET2_MULTITHREADING:BOOL=ON -DBUILD_SHARED_LIBS:BOOL=ON -DBUILD_UNIT_TESTS:BOOL=OFF`

Levi-Armstrong, 2022-01-21:
These get added where the targets are created. The first goes in the `bullet/CMakelists.txt` and the second goes in `vhacd/CMakelists.txt`. You will need to install the latest cmake using pip.

Levi-Armstrong, 2022-01-26:
@Davidlequnchen Did this change resolve the issue for you on 18.04?

Yuki-cpp, 2022-02-17:
Hello,

I was having a very similar issue and the fix you provided (adding a `target_link_directories` for the bullet and vhcd targets) worked for me.

For future reference though, I had to add a scope in each command:
```cmake
target_link_directories(${PROJECT_NAME}_bullet PUBLIC "${BULLET_ROOT_DIR}/${BULLET_LIBRARY_DIRS}")
# And
target_link_directories(${PROJECT_NAME}_vhacd PUBLIC "${BULLET_ROOT_DIR}/${BULLET_LIBRARY_DIRS}")
```
Tested with `cmake 3.15.0` and should work like that with anything above `3.13`

Davidlequnchen, 2022-02-17:
> @Davidlequnchen Did this change resolve the issue for you on 18.04?

Sorry for the late reply. 
Yes, this resolved my issue by adding the ```target_link_directories``` for the bullet and vhcd targets on Ubuntu 18.04.
Thanks a lot for the help!

Btw, I got an error showing GTEST was not properly installed
```
Could NOT find GTest (missing: GTEST_LIBRARY GTEST_MAIN_LIBRARY)
```
solved it by: https://stackoverflow.com/questions/24295876/cmake-cannot-find-googletest-required-library-in-ubuntu

closed 2022-02-17 by Davidlequnchen