tesseract-robotics/tesseract #700: Ubuntu 20.04 ROS Noetic build error: no matching function for call to ‘checkTrajectorySegment

Davidlequnchen, 2022-01-21:
As mentioned earlier in #699, I got build errors for Ubuntu 18.04 ROS Melodic. So I installed a virtual machine for Ubuntu 20.04 this time with ROS noetic to build the whole workspace.

The build log:
```text

Workspace configuration appears valid.
-------------------------------------------------------------------------------------------------------------------------
[build] Found '31' packages in 0.0 seconds.                                                                             
[build] Package table is up to date.                                                                                    
Starting  >>> descartes_light                                                                                           
Starting  >>> ifopt                                                                                                     
Starting  >>> osqp                                                                                                      
Starting  >>> tesseract_common                                                                                          
Starting  >>> tesseract_msgs                                                                                            
Starting  >>> trajopt_utils                                                                                             
Finished  <<< ifopt                                          [ 0.5 seconds ]                                            
Finished  <<< trajopt_utils                                  [ 0.3 seconds ]                                            
Finished  <<< tesseract_common                               [ 0.3 seconds ]                                            
Starting  >>> tesseract_command_language                                                                                
Starting  >>> tesseract_support                                                                                         
Starting  >>> vhacd                                                                                                     
Finished  <<< descartes_light                                [ 0.3 seconds ]                                            
Finished  <<< tesseract_msgs                                 [ 0.9 seconds ]                                            
Finished  <<< osqp                                           [ 1.5 seconds ]                                            
Starting  >>> osqp_eigen                                                                                                
Starting  >>> trajopt_sco                                                                                               
Finished  <<< tesseract_command_language                     [ 0.3 seconds ]                                            
Starting  >>> tesseract_time_parameterization                                                                           
Finished  <<< tesseract_support                              [ 0.2 seconds ]                                            
Starting  >>> tesseract_geometry                                                                                        
Finished  <<< vhacd                                          [ 0.3 seconds ]                                            
Finished  <<< trajopt_sco                                    [ 0.3 seconds ]                                            
Finished  <<< tesseract_time_parameterization                [ 0.3 seconds ]                                            
Finished  <<< tesseract_geometry                             [ 0.3 seconds ]                                            
Starting  >>> tesseract_scene_graph                                                                                     
Finished  <<< osqp_eigen                                     [ 0.8 seconds ]                                            
Finished  <<< tesseract_scene_graph                          [ 0.3 seconds ]                                            
Starting  >>> tesseract_collision                                                                                       
Starting  >>> tesseract_srdf                                                                                            
Finished  <<< tesseract_collision                            [ 0.6 seconds ]                                            
Starting  >>> tesseract_urdf                                                                                            
Finished  <<< tesseract_srdf                                 [ 0.3 seconds ]                                            
Finished  <<< tesseract_urdf                                 [ 0.4 seconds ]                                            
Starting  >>> tesseract_state_solver                                                                                    
Finished  <<< tesseract_state_solver                         [ 0.3 seconds ]                                            
Starting  >>> tesseract_kinematics                                                                                      
Finished  <<< tesseract_kinematics                           [ 0.5 seconds ]                                            
Starting  >>> tesseract_environment                                                                                     
Finished  <<< tesseract_environment                          [ 0.3 seconds ]                                            
Starting  >>> tesseract_visualization                                                                                   
Finished  <<< tesseract_visualization                        [ 0.3 seconds ]                                            
Starting  >>> trajopt                                                                                                   
________________________________________________________________________________________________________________________
Warnings   << trajopt:check /home/lequn/tesseract_ws/logs/trajopt/build.check.000.log                                   
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  CMake variable PCL_ROOT is set to:

    /usr

  For compatibility, CMake is ignoring the variable.
Call Stack (most recent call first):
  /home/lequn/tesseract_ws/devel/lib/cmake/tesseract_urdf/tesseract_urdf-config.cmake:29 (find_dependency)
  /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
[... 23 lines cut ...]
      |                                                                                                            ^
In file included from /home/lequn/tesseract_ws/src/tesseract_planning/tesseract_motion_planners/core/src/core/utils.cpp:35:
/home/lequn/tesseract_ws/devel/include/tesseract_environment/utils.h:62:39: note: candidate: ‘tesseract_collision::ContactResultMap tesseract_environment::checkTrajectorySegment(tesseract_collision::ContinuousContactManager&, const TransformMap&, const TransformMap&, const tesseract_collision::CollisionCheckConfig&)’
   62 | tesseract_collision::ContactResultMap checkTrajectorySegment(tesseract_collision::ContinuousContactManager& manager,
      |                                       ^~~~~~~~~~~~~~~~~~~~~~
/home/lequn/tesseract_ws/devel/include/tesseract_environment/utils.h:62:39: note:   candidate expects 4 arguments, 5 provided
/home/lequn/tesseract_ws/devel/include/tesseract_environment/utils.h:77:1: note: candidate: ‘tesseract_collision::ContactResultMap tesseract_environment::checkTrajectorySegment(tesseract_collision::ContinuousContactManager&, const TransformMap&, const TransformMap&, const tesseract_collision::ContactRequest&)’
   77 | checkTrajectorySegment(tesseract_collision::ContinuousContactManager& manager,
...
........................................................................................................................
Failed     << tesseract_motion_planners:make                 [ Exited with code 2 ]                                     
[... cut ...]
[build]   Ignored:   None.                                                                                              
[build]   Warnings:  3 packages succeeded with warnings.                                                                
[build]   Abandoned: 6 packages were abandoned.                                                                         
[build]   Failed:    1 packages failed.                                                                                 
[build] Runtime: 17.8 seconds total.
```

Levi-Armstrong, 2022-01-21:
The master branch can be out of sink as we get PR reviewed so I recommend pulling the latest tags on tesseract and tesseract_planning.

Davidlequnchen, 2022-01-21:
> The master branch can be out of sink as we get PR reviewed so I recommend pulling the latest tags on tesseract and tesseract_planning.

Thanks a lot. I have succesfully build the workspace with previous tages.

closed 2022-01-21 by Davidlequnchen