tesseract-robotics/tesseract #706: Add ability to search for all plugins of specific type

marip8, 2022-01-28:
Currently the each `tesseract` plugin implementation exports a unique symbol which generally corresponds to the name of the implementation class. A specific plugin is loaded by specifying its unique symbol name. This makes it difficult for an application that doesn't know these unique names (such as a GUI) to query for all available plugins (e.g. inverse kinematics, contact managers, visualization, planners, etc.) that match a particular interface. 

It looks like the `boost` solution for this is to use the `BOOST_DLL_ALIAS_SECTIONED` macro with a "section" name when exporting the plugin. A common section name serves as a namespace for multiple plugins exported with different symbol names.

One solution to this problem would be to update the `TESSERACT_ADD_PLUGIN` macro from 

https://github.com/tesseract-robotics/tesseract/blob/e9d6f064f14c84bb14c65c2ebfe529a325133a80/tesseract_common/include/tesseract_common/class_loader.h#L93-L95

to: 

```c++
#define TESSERACT_ADD_PLUGIN(DERIVED_CLASS, ALIAS, SECTION) \ 
  BOOST_DLL_ALIAS_SECTIONED(DERIVED_CLASS, ALIAS, SECTION)
```
where `SECTION` could be the name of the abstract base class (e.g. `ContactManager`, `InverseKinematics`, etc.) or something similar. A method could then be added to the plugin loader class to list all symbol names under a given section name per [this tutorial](https://www.boost.org/doc/libs/1_70_0/doc/html/boost_dll/tutorial.html#boost_dll.tutorial.querying_libraries_for_symbols). 

An alternative would be to export all plugin implementations with the same alias (which would be something like the abstract base class name) and build them in separate libraries. `boost` provides an example of that [here](https://www.boost.org/doc/libs/1_70_0/doc/html/boost_dll/tutorial.html#boost_dll.tutorial.symbol_shadowing_problem__linux_). However, this method would require us to split up the compilation of several existing plugins that are currently in the same library (e.g. contact managers and possibly others)

Levi-Armstrong, 2022-01-28:
I like the section approach.

marip8, 2022-01-28:
Should be a pretty minimally invasive change. I might be able to tackle this next week.

Levi-Armstrong, 2022-05-27:
This has been addressed by PR #741

closed 2022-05-27 by Levi-Armstrong