tesseract-robotics/tesseract #707: tesseract_geometry example build failure

Briancbn, 2022-01-28:
**OS:** Ubuntu 20.04
**ROS Distro:** Foxy
**Branch:** Source build from master `0.8.2`

I don't know why this doesn't show up in the GHA here.

```
--- stderr: tesseract_geometry
CMake Error at examples/CMakeLists.txt:16 (add_executable):
  Target "tesseract_geometry_parse_mesh_example" links to target
  "tesseract::tesseract_common" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
CMake Error at examples/CMakeLists.txt:16 (add_executable):
  Target "tesseract_geometry_parse_mesh_example" links to target
  "tesseract::tesseract_support" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
CMake Error at examples/CMakeLists.txt:3 (add_executable):
  Target "tesseract_geometry_create_geometries_example" links to target
  "tesseract::tesseract_common" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
CMake Error at examples/CMakeLists.txt:3 (add_executable):
  Target "tesseract_geometry_create_geometries_example" links to target
  "tesseract::tesseract_support" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

Levi-Armstrong, 2022-01-29:
@Briancbn by chance are you using the latest ros_industrial_cmake_boilerplate?

Levi-Armstrong, 2022-01-29:
@Briancbn A bug was introduce in ros_industrial_cmake_boilerplate which has been addressed by this [PR](https://github.com/ros-industrial/ros_industrial_cmake_boilerplate/pull/60). Let me know if it resolves the issue.

Briancbn, 2022-01-30:
Thanks @Levi-Armstrong for the prompt reply and fix, that solves the issue!

closed 2022-01-30 by Briancbn