tesseract-robotics/tesseract #712: UR parameters seem to be slightly off

Yuki-cpp, 2022-02-17:
Hello,

In [tesseract_kinematics/core/types.h](https://github.com/tesseract-robotics/tesseract/blob/2673ea6556a887520615bea38d10a557444353b5/tesseract_kinematics/core/include/tesseract_kinematics/core/types.h#L60), the UR Parameters for each robot are defined as follow:

```cpp
const static URParameters UR10Parameters(0.1273, -0.612, -0.5723, 0.163941, 0.1157, 0.0922);
const static URParameters UR5Parameters(0.089159, -0.42500, -0.39225, 0.10915, 0.09465, 0.0823);
const static URParameters UR3Parameters(0.1519, -0.24365, -0.21325, 0.11235, 0.08535, 0.0819);
const static URParameters UR10eParameters(0.181, -0.613, -0.571, 0.135 + 0.176 - 0.137, 0.120, 0.117);
const static URParameters UR5eParameters(0.163, -0.425, -0.392, 0.127 + 0.138 - 0.131, 0.100, 0.100);
const static URParameters UR3eParameters(0.152, -0.244, -0.213, 0.104 + 0.138 - 0.131, 0.085, 0.092);
```

However, when looking at [official documentation](https://www.universal-robots.com/articles/ur/application-installation/dh-parameters-for-calculations-of-kinematics-and-dynamics/), we can notice that the values are slightly different. For example, UR5e values should be:
```cpp
const static URParameters UR5eParameters(0.1625, -0.425, -0.3922, 0.1333, 0.0997, 0.0996);
//Where the current ones when the math is done are
const static URParameters UR5eParameters(0.163, -0.425, -0.392, 0.134, 0.100, 0.100);
```
Those differences are pretty small, they can add up (Initial testing gave us a ~1mm error with an UR5e). 
Is there any reason for those values to be rounded as they are now? If not, and if you wish so, I can submit a PR to provide a fix (it might also be the occasion  to add UR16e to the supported robots on that solver).

gavanderhoorn, 2022-02-17:
If you're really concerned about precision, using the ideal values from the page you link would not be sufficient.

You'd have to use the exact calibration for the UR you're using.

The `Universal_Robots_ROS_Driver` is capable of extracting that calibration data and serialising it to a `.yaml` file.

Levi-Armstrong, 2022-02-17:
I believe these numbers were pulled from the UR repository urdf data. If you want to create a PR to update them that would be appreciated. Also the kinematics are loaded as plugins so you could also do as @gavanderhoorn suggested and then update the parameters in the [kinematics plugin yaml](https://tesseract-docs.readthedocs.io/en/latest/_source/core/packages/tesseract_kinematics_doc.html). Now this only updates the inverse kinematics so the forward kinematics from the URDF would also need to be updated which I believe is supported by the UR xacro.

Yuki-cpp, 2022-02-18:
Thanks a lot for the pointers about loading the actual values. i'll have a look at that.
In the meantime I submitted a quick PR to update the values to match.

Yuki-cpp, 2022-02-21:
PR got merged so I guess I can close this one :)

closed 2022-02-21 by Yuki-cpp