tesseract-robotics/tesseract #731: Improve windows CI build speed.

Levi-Armstrong, 2022-03-28:
@johnwason Hey do you have any ideas on how we can speed up the windows CI build?

johnwason, 2022-03-28:
The only simple ways to speed up the build is to use caching and disable build optimizations, which are pretty slow. If you fix the osqp issue with static and shared libs having the same name you can use ninja, which may also be a bit faster.

Levi-Armstrong, 2022-03-28:
It looks like the osqp has been resolved but it is in the develope-1.0 branch. I asked a question on when these will get merged into master.

johnwason, 2022-04-15:
I noticed patching Bullet that it builds a lot of unit tests and other junk that we don't need. There may be more options that can be disabled so it builds faster.

Levi-Armstrong, 2022-06-29:
@johnwason I switched to how you are building your repositories on windows using vcpkg and caching the depends which has improved the windows CI build.

closed 2022-06-29 by Levi-Armstrong