tesseract-robotics/tesseract #733: Compiling problem

RobotEmperor, 2022-03-28:
Hello, tesseract developer team

I was in trouble about compiling tesseract on ROS. 
When I complied with the version from 0.1.0 to 0.6.0, it worked fine. 
However, I tried to use the above version, I have this problem:
[image]
I installed everything including libbullet-dev.
Do you have any suggestions to resolve this issue?

Levi-Armstrong, 2022-03-28:
Are you on 18.04 or 20.04?

RobotEmperor, 2022-03-28:
Sorry for the confusion. I'm using Ubuntu 20.04 + ROS noetic.

Levi-Armstrong, 2022-03-28:
Did you also install `libbullet-extras-dev`?

Levi-Armstrong, 2022-03-28:
Also if you building from source I would make sure you don't have it installed from ros. There is a known issue which we are working on resolving so I would run `sudo apt remove ros-noetic-tesseract-*`

RobotEmperor, 2022-03-28:
Thank you for your ideas. I also installed libbullet-extras-dev. When I tried to remove the ros-noetic-tesseract-*, I didn't see the package. I think the package isn't installed before. I've tried to figure it out more. Could you have another suggestion?

Levi-Armstrong, 2022-03-29:
Are you using a recent tag? The issue is it is not finding bullet built with double precision. 

Do you have this file on your machine? It may be in a different location.

`/usr/lib/x86_64-linux-gnu/cmake/bullet/BulletConfig-float64.cmake`

RobotEmperor, 2022-03-29:
I tried with the master version, it works fine. Thank you for your help.

closed 2022-03-29 by RobotEmperor