tesseract-robotics/tesseract #757: How to use IKFast for new robot type?

fishbotics, 2022-04-26:
Hi all,

I'm just getting started using Tesseract and I'd like to be able to use it for the Franka Panda. In the past, I've used IKFast (because, well, it's  fast), but I had to generate a new C++ file using OpenRave. In the ikfast directory, I don't see any robot-specific info. Can someone point me to how I might be able to integrate a new robot type into the IK solver?

Thanks!

Levi-Armstrong, 2022-04-26:
Take a look [here](https://github.com/tesseract-robotics/tesseract/tree/master/tesseract_kinematics/test) at the following files.
- abb_irb2400_ikfast_kinematics.h
- abb_irb2400_ikfast_kinematics.cpp
- abb_irb2400_ikfast_solver.hpp.

Also there is iiwa7 example in the same directory.

Levi-Armstrong, 2022-04-26:
After you create the kinematics object you will need to create the [factory](https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_kinematics/kdl/src/kdl_factories.cpp) which is the plugin to tesseract.

fishbotics, 2022-05-05:
[image]

It looks pretty similar to the OPW model, but I don't know how to get all the parameters. Maybe I can load it in sim and figure out how to take the measurements.

Levi-Armstrong, 2022-05-09:
> Thanks @Levi-Armstrong. I might be misunderstanding here, but that factory you linked is for KDL. I see in [here](https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_kinematics/test/kinematic_plugins.yaml) that you list all the plugins. The iiwa seems to point to KDL--not IKFast and abb points to OPW, which I think is defined [here](https://github.com/tesseract-robotics/tesseract/blob/8445d507a2bbe3ec5b5e819a835a0d33758cc13b/tesseract_kinematics/opw/include/tesseract_kinematics/opw/opw_factory.h).
> 
> Is the OPW IK library your preferred standard? I'm used to using IKFast, but I'd be happy to set up something else if it's better. Or, do you have an example of an IKFast factory?

@fishbotics Here are the ikfast examples for the iiwa

https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_kinematics/test/iiwa7_ikfast_kinematics.h
https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_kinematics/test/iiwa7_ikfast_kinematics.cpp

Generated from ikfast
https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_kinematics/test/iiwa7_ikfast_solver.hpp

marip8, 2022-05-09:
OPW kinematics can only be used with a very specific robot configuration, and I don't think this particular robot meets the criteria. A few general requirements are that the robot must be 6 DOF and have a spherical wrist (i.e. the last three axes all intersect at a single point). [Here is a picture of a valid OPW configuration with the dimensions denoted](https://github.com/Jmeyer1292/opw_kinematics/blob/3c66f52b26279cf04c157efe15d8c37ada8e49c1/opw.png). If your robot doesn't have an exact correlation to this diagram, then OPW can't be used

marip8, 2022-05-13:
I'm coincidentally also trying to implement a custom IKFast plugin currently, and it seems like the functionality for loading a custom plugin is fairly limited in one case and broken in another. 

## Case 1: YAML config file (limited capability)
You can specify the names and locations of plugins in the plugin config file, such as [this example](https://github.com/tesseract-robotics/tesseract/blob/eebe3fd4570bf4c4a0cfc548ce0299d6b2a3c3df/tesseract_kinematics/test/kinematic_plugins.yaml#L2-L5). However, I believe these names have to be hard-coded, so they wouldn't transfer between machines unless the plugins were installed to a universal directory (which is not the case for most ROS users building in a workspace in their home directory).

## Case 2: Environment variables (broken)
You can also specify the names and locations of plugins using [two (fairly hidden) environment variables](https://github.com/tesseract-robotics/tesseract/blob/eebe3fd4570bf4c4a0cfc548ce0299d6b2a3c3df/tesseract_kinematics/core/src/kinematics_plugin_factory.cpp#L35-L36). A user could use a script or ROS launch file to populate these environment variables in a generic way that is machine independent.

However, the plugin loader does not attempt to load plugins that it finds from environment variables. Instead it only loads libraries from its internal cache (see below that `plugins_local` never really gets used). I think the fix for this is simple: iterate over `plugins_local` rather than `this->search_libraries`.

https://github.com/tesseract-robotics/tesseract/blob/eebe3fd4570bf4c4a0cfc548ce0299d6b2a3c3df/tesseract_common/include/tesseract_common/plugin_loader.hpp#L133-L162

Are there other ways of loading custom plugins, or do we need to fix these methods? 

Also the kinematic plugin unit tests don't test this loading capability because the plugin loader already knows about the plugins built by `tesseract` through the [`TESSERACT_KINEMATICS_PLUGINS` compiler definition](https://github.com/tesseract-robotics/tesseract/blob/eebe3fd4570bf4c4a0cfc548ce0299d6b2a3c3df/tesseract_kinematics/core/src/kinematics_plugin_factory.cpp#L50-L51).

marip8, 2022-05-13:
I guess there is a third case which is to use some combination of both other methods, specifically setting the search path in the environment variable and adding the plugin library name to the YAML config file. We still should address the shortcomings of the other approaches too though

Levi-Armstrong, 2022-05-13:
> ## Case 1: YAML config file (limited capability)
> You can specify the names and locations of plugins in the plugin config file, such as [this [...]

I don't think that is the case. It first searches the directories but then reverts to system location and what is in your environment variables. I usually don't add anything here and as long as the terminal is sourced it will find them.

Levi-Armstrong, 2022-05-13:
> ## Case 2: Environment variables (broken)
> You can also specify the names and locations of plugins using [two (fairly hidden) environment [...]

The environment variable was added to support containerization like using snap. Other than that it should really never need to be used because it searches the system location for the library.

Levi-Armstrong, 2022-05-13:
> However, the plugin loader does not attempt to load plugins that it finds from environment variables. Instead it only loads libraries from its internal cache (see below that plugins_local never [...]

This piece may very well be broke and will need to look at it more closely.

Levi-Armstrong, 2022-05-13:
@marip8 Do you run into a case where it was unable to find the plugin?

marip8, 2022-05-13:
> > ## Case 1: YAML config file (limited capability)
> > You can specify the names and locations of plugins in the plugin config file, such as [this example](https://github.com/tesseract-robotics/tesseract/blob/eebe3fd4570bf4c4a0cfc548ce0299d6b2a3c3df/tesseract_kinematics/test/kinematic_plugins.yaml#L2-L5). However, I believe these names have to be hard-coded, so they wouldn't transfer between machines unless the plugins were installed to a universal directory (which is not the case for most ROS users building in a workspace in their home directory).
> 
> I don't think that is the case. It first searches the directories but then reverts to system location and what is in your environment variables. I usually don't add anything here and as long as the terminal is sourced it will find them.
> 
> Update: To clarify, you do need to add the library to the list but really do not need to add a search path.

For some reason this doesn't seem to work on a ROS2 system after I have sourced my workspace. It shows a search path in my tools workspace where `tesseract` was built, but the current workspace where the custom plugin is built does not have a known search path

```
snp_motion_planning_node-6] Warning: Failed to load symbol 'MotomanHC10InvKinFactory'
[snp_motion_planning_node-6]          at line 310 in /home/mripperger/ros/snpd/snpd_tools_ws/src/tesseract/tesseract_kinematics/core/src/kinematics_plugin_factory.cpp
[snp_motion_planning_node-6] Error:   Failed to instantiate plugin 'MotomanHC10InvKinFactory', Details: 
[snp_motion_planning_node-6] Search Paths (Search System Folders: True):
[snp_motion_planning_node-6]     - /home/mripperger/ros/snpd/snpd_tools_ws/install/tesseract_kinematics/lib
[snp_motion_planning_node-6] Search Libraries:
[snp_motion_planning_node-6]     - libsnp_motion_planning_plugins.so
[snp_motion_planning_node-6]     - libtesseract_kinematics_core_factories.so
[snp_motion_planning_node-6]     - libtesseract_kinematics_kdl_factories.so
[snp_motion_planning_node-6]     - libtesseract_kinematics_opw_factory.so
[snp_motion_planning_node-6]     - libtesseract_kinematics_ur_factory.so

```

Levi-Armstrong, 2022-05-14:
Is `libsnp_motion_planning_plugins.so` location in your LD_LIBRARY_PATH environment variable?

Levi-Armstrong, 2022-05-14:
I also think it would be good to update the parsing to of search path to use the resource locator if possible so you may use a package url.

Levi-Armstrong, 2022-05-14:
Here is the documentation for what [dlopen](https://man7.org/linux/man-pages/man3/dlopen.3.html) searches.

Levi-Armstrong, 2022-05-14:
@marip8 Is this library by chance in a pure cmake package?

marip8, 2022-05-16:
> Is this library by chance in a pure cmake package?

No, it's in a ROS2 package.

> Is libsnp_motion_planning_plugins.so location in your LD_LIBRARY_PATH environment variable?

Actually no. Turns out it was a few ROS2 errors. I forgot to add an `ament_export_targets` command, so the package location wasn't added to the `LD_LIBRARY_PATH`. I also didn't realize that ROS2 libraries need to be installed in `lib` not `lib/<package_name>`. Once I fixed those, it works as expected