tesseract-robotics/tesseract #764: Build failure on Ubuntu 22.04

johnwason, 2022-05-24:
I had a strange build failure on Ubuntu 22.04:

```
In file included from /usr/include/boost/serialization/unordered_map.hpp:27,
                 from /__w/tesseract_python-1/tesseract_python-1/ws/src/tesseract/tesseract_common/src/allowed_collision_matrix.cpp:31:
/usr/include/boost/serialization/unordered_collections_load_imp.hpp: In function ‘void boost::serialization::stl::load_unordered_collection(Archive&, Container&)’:
/usr/include/boost/serialization/unordered_collections_load_imp.hpp:51:27: error: ‘library_version_type’ is not a member of ‘boost::serialization’; did you mean ‘item_version_type’?
   51 |     boost::serialization::library_version_type library_version(
      |                           ^~~~~~~~~~~~~~~~~~~~
      |                           item_version_type
/usr/include/boost/serialization/unordered_collections_load_imp.hpp:57:30: error: ‘library_version_type’ is not a member of ‘boost::serialization’; did you mean ‘item_version_type’?
   57 |     if(boost::serialization::library_version_type(3) < library_version){
      |                              ^~~~~~~~~~~~~~~~~~~~
      |                              item_version_type
/usr/include/boost/serialization/unordered_collections_load_imp.hpp:57:56: error: ‘library_version’ was not declared in this scope
   57 |     if(boost::serialization::library_version_type(3) < library_version){
      |                                                        ^~~~~~~~~~~~~~~
```
https://github.com/johnwason/tesseract_python-1/runs/6580719265?check_suite_focus=true#step:8:618

I don't know if this is a problem with Boost or with tesseract_common.

johnwason, 2022-05-25:
It looks like

```
#include <boost/serialization/library_version_type.hpp>
```

is missing for the version of boost serialization in 22.04.

johnwason, 2022-05-26:
I am seeing other unrelated build failures in other files.

rjoomen, 2022-06-04:
Seem to be [this issue](https://github.com/boostorg/serialization/issues/219) in Boost 1.74, which is the default for Ubuntu 22.04

closed 2022-09-04 by Levi-Armstrong