tesseract-robotics/tesseract #766: Potential unit test sensitivity issue

dpiet, 2022-05-27:
While troubleshooting kinematics issues with a different ABB URDF implementation, I encountered a failing unit test in tesseract_kinematics at https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_kinematics/test/kinematics_core_unit.cpp#L181. The `UtilscalcManipulabilityUnit` test passes for the included ABB IRB2400 URDF. But when I replace the URDF with a different ABB IRB robot in the `getSceneGraphABB()` call, the `calcManipulability` method is returning `-NaN` for volume and this breaks the unit test EXPECT_NEAR condition. Same with some of the other checks that involve the eigenvalue data.

Values for test that passes with ABB IRB2400:
```
jacobian:
      0   0.84  0.135      0      0      0
  0.94      0      0      0      0      0
     0  -0.84  -0.84      0 -0.085      0
     0      0      0      1      0      1
     0      1      1      0      1      0
     1      0      0      0      0      0
eigenvalues:
     4.56009
   0.441406
   0.140754
     1.8836
3.64023e-18
          2
Volume: 1.9711e-09
```

Values for failing test with a different ABB IRB URDF:
```
jacobian:
      0  1.075  0.175      0      0      0
  1.27      0      0      0      0      0
     0 -1.095 -1.095      0 -0.135      0
     0      0      0      1      0      1
     0      1      1      0      1      0
     1      0      0      0      0      0
eigenvalues:      
    5.79023
    0.595964
    0.216327
      2.6129
-2.73947e-16
           2
Volume: -nan
```
Eigen library version 3.3.7

Could this be an Eigen output issue or something else?

Levi-Armstrong, 2022-05-27:
@dpiet Would you be willing to test this [PR](https://github.com/tesseract-robotics/tesseract/pull/768) to see if it resolves the issue?

dpiet, 2022-05-31:
Hi @Levi-Armstrong I was just able to test the fix this morning and looks good. Thanks for the merge.