tesseract-robotics/tesseract #785: fatal error: tesseract_common/type_erasure.h: No such file or directory

fishbotics, 2022-06-27:
Hi all,

I'm trying to build following the tutorial here: https://tesseract-docs.readthedocs.io/en/latest/_source/intro/getting_started_doc.html

When I get to the build stage, I am getting an error saying that there is no type_erasure.h file. I followed all the steps in the instructions. The only deviations were:

1. I didn't clone `tesseract_ros`
2. I had to manually clone `ros_industrial_cmake_boilerplate` into my `src` directory because rosdep didn't seem to install this
3. I had to manually install lcov

Here is an image of the error message:
[image]

Thanks a lot!

fishbotics, 2022-06-28:
Digging into this further, I see that these files are missing in `/opt/ros/noetic/include/tesseract_common`. Maybe one of the steps to install the dependencies is cloning an old version that doesn't match `master` on `tesseract_planning`? I'll try to find the root cause and post it.

Levi-Armstrong, 2022-06-28:
What tag do you have for tesseract repository?

fishbotics, 2022-06-28:
I have been following the commands in that link, which had me run `git clone https://github.com/tesseract-robotics/tesseract_planning`, which is currently set at https://github.com/tesseract-robotics/tesseract_planning/commit/75cd1a305d2ce4d18a6912e616efaba281c316e4

fishbotics, 2022-06-28:
For what it's worth, I tried doing the same thing using v0.11.0 (cloned `tesseract_planning` and checked out https://github.com/tesseract-robotics/tesseract_planning/commit/4ec9be044477cd3f0f699df89480a78f345cad19 before running the other steps. This time, I did not need to clone the cmake tools repo, but I still did need to install `lcov`. Afterward, the error message still appears.

FWIW, I'm building in a clean docker container each time, so there isn't any carryover between installations.

@Levi-Armstrong , what version do you suggest I build with?

Levi-Armstrong, 2022-06-28:
I would clone the latest tags on tesseract, tesseract_planning, tesserect_qt and tesseract_ros if you need the ros components.

fishbotics, 2022-06-28:
OK so should I avoid installing dependencies via `wstool init ~/tesseract_ws/src/ ~/tesseract_ws/src/tesseract_planning/dependencies.rosinstall` and `rosdep install -y --from-paths ~/tesseract_ws/src --ignore-src --rosdistro noetic`? I think one of those must be downloading the older version of `tesseract_common`

fishbotics, 2022-06-28:
I am no expert in `rosdep`, but maybe the version is behind? I see this version number (0.8.6-1): https://github.com/ros/rosdistro/blob/master/noetic/distribution.yaml#L10262

Meanwhile, when I look at that version in the main repo, the necessary file is missing: https://github.com/tesseract-robotics/tesseract/tree/0.8.6-1/tesseract_common/include/tesseract_common

Levi-Armstrong, 2022-06-28:
Oh, okay. I would run the following command `sudo apt remove ros-noetic-tesseract*` and then clone the repositories into your workspace. There is a known issue with having the system version installed when trying to build from source. I will be pushing a  release soon which hopefully solves the issue.

fishbotics, 2022-06-28:
Alright I got it to build. Just for documentation purposes, here are the steps I took (built from a clean docker with the full desktop version of ros noetic already installed, along with catkin tools).

```
mkdir -p tesseract_ws/src
cd tesseract_ws/src
git clone https://github.com/tesseract-robotics/tesseract_planning.git && cd tesseract_planning && git checkout tags/0.11.0 -b 0.11.0 && cd -
wstool init ~/tesseract_ws/src/ ~/tesseract_ws/src/tesseract_planning/dependencies.rosinstall
git clone https://github.com/ompl/ompl.git
git clone https://github.com/Jmeyer1292/opw_kinematics.git
git clone https://github.com/flexible-collision-library/fcl.git
git clone https://github.com/ros-industrial/ros_industrial_cmake_boilerplate && cd ros_industrial_cmake_boilerplate && git checkout tags/0.2.16 -b 0.2.16 && cd -
git clone https://github.com/tesseract-robotics/tesseract_ros.git && cd tesseract_ros && git checkout tags/0.5.1 -b 0.5.1 && cd -
git clone https://github.com/tesseract-robotics/tesseract_qt.git && cd tesseract_qt && git checkout tags/0.1.2 -b 0.1.2 && cd -
git clone https://github.com/tesseract-robotics/tesseract.git && cd tesseract && git checkout tags/0.9.10 -b 0.9.10 && cd -
apt install lcov libbullet-extras-dev coinor-libipopt-dev ros-noetic-octomap-msgs
catkin build
```

Levi-Armstrong, 2022-06-28:
The qt_advanced_docking_system can be installed via `sudo apt install ros-noetic-qt-advanced-docking`
you should not need to install build the following from source. This could be causing issues because it is the latest which I have been using the system version for these.
```
git clone https://github.com/ompl/ompl.git
git clone https://github.com/Jmeyer1292/opw_kinematics.git
git clone https://github.com/flexible-collision-library/fcl.git
```

Levi-Armstrong, 2022-06-28:
I would also follow this procedure.

``` bash
mkdir -p tesseract_ws/src
cd tesseract_ws/src
git clone -b master https://github.com/tesseract-robotics/tesseract_planning.git 
git clone -b 0.2.16 https://github.com/ros-industrial/ros_industrial_cmake_boilerplate.git
git clone -b master https://github.com/tesseract-robotics/tesseract_ros.git
git clone -b main https://github.com/tesseract-robotics/tesseract_qt.git
git clone -b master https://github.com/tesseract-robotics/tesseract.git 
cd ..
rosdep install --from-paths src --ignore-src -r -y -q
catkin build
```

fishbotics, 2022-06-30:
@Levi-Armstrong , this is working now, so I'll go ahead and close it up. Thanks for your help. Might be worth mentioning this issue again when you update the release.

closed 2022-06-30 by fishbotics

emfebert, 2023-08-16:
@Levi-Armstrong  After running `rosdep install --from-paths src --ignore-src -r -y -q`

I'm getting

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
tesseract_qt: Cannot locate rosdep definition for [gz-math7]
tesseract_motion_planners: Cannot locate rosdep definition for [trajopt_sqp]
tesseract_examples: Cannot locate rosdep definition for [trajopt_ifopt]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```