tesseract-robotics/tesseract #798: URDF Exporter URI support

marip8, 2022-07-08:
It looks like the URDF writing capabilities of `tesseract_urdf` assume that all geometry is referenced with the `package://` URI in the output URDF. This is not necessarily always the case, especially with non-ROS projects that use tesseract to generate a URDF model. Has there been any thought behind supporting other URIs, specifically `file://`?

Levi-Armstrong, 2022-07-09:
The default resource locators support this so it makes sense that they would support it also.

JeremyZoss, 2022-07-11:
So what does that argument to the export method look like?  An enumeration of the possible export types?  An `export_uri_prefix` string that gets prepended to the mesh file locations?  Do we actively try to write the files to the location specified by the resource prefix, or do we always write the geometry to the same place and assume the user specifies the right prefix for where they plan to place the files?

[It looks like](https://github.com/ros/resource_retriever/blob/kinetic-devel/package.xml#L9-L14) the default locator uses libcurl for imports, and supports many formats: http, ftp, file, package, etc.  The full [libcurl](https://curl.se/libcurl/) format list is pretty extensive.  You could maybe wrap libcurl to copy the files to the specified location.  If we're only prepending a user-provided resource location prefix (not actually copying files to the specified location), then this is not really an issue.

Also note that it seems like `file://` URIs always expect absolute paths (not relative paths).  So I'm not sure if there's a good portable method to replace the standard package-relative resource reference with something less ROS-centric.

Levi-Armstrong, 2022-07-12:
Maybe we create a base class called ResourceExporter with a virtual method given a file name generates a URI. Then all of the write methods in tesseract_urdf take this instead of package_path and file name. This way the developer could implement there own way of generating URI using file://, package, etc.