tesseract-robotics/tesseract #817: /usr/bin/ld: tesseract_geometry_serialization_unit: hidden symbol `__gcov_init' in /usr/lib/gcc/x86_64-linux-gnu/9/libgcov.a(_gcov.o) is referenced by DSO

dbdxnuliba, 2022-09-20:
/usr/bin/ld: tesseract_geometry_serialization_unit: hidden symbol `__gcov_init' in /usr/lib/gcc/x86_64-linux-gnu/9/libgcov.a(_gcov.o) is referenced by DSO
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/tesseract_geometry_serialization_unit.dir/build.make:99: test/tesseract_geometry_serialization_unit] Error 1
make[1]: *** [CMakeFiles/Makefile2:461: test/CMakeFiles/tesseract_geometry_serialization_unit.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

my enviroment is gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1)
ubuntu20.04 noetic

Levi-Armstrong, 2022-09-24:
Do you have gcov installed?

Levi-Armstrong, 2022-09-25:
This is address by PR #819

closed 2022-09-25 by Levi-Armstrong