tesseract-robotics/tesseract #828: Octomap segfault

marip8, 2022-10-19:
Here is a [PR](https://github.com/OctoMap/octomap/pull/288/files#diff-45151bec6492789e0b7d1128eb0a8488442d65ccc2756d631d68219c2a8cd180) which fixes the issue. I think we need to add the same logic to the pruneNode in tesseract_geometry

_Originally posted by @Levi-Armstrong in https://github.com/tesseract-robotics/tesseract_planning/issues/255#issuecomment-1278024415_

marip8, 2022-10-19:
Per @Levi-Armstrong 

> I tried to add the following in the prune function, but children is not accessible.
> 
> ```
>         if (node->children != NULL){
>           delete[] node->children;
>           node->children = NULL;
>         }
> ```

Levi-Armstrong, 2022-10-20:
I am starting to think there may be an issue with it finding different version of the library. The system installs version 1.9.3 and ros noetic has its own version of 1.9.8. If you run ldd on the example library and executable which is creating the error do you only see it linked against one version?

marip8, 2023-03-09:
> I am starting to think there may be an issue with it finding different version of the library.

This is for sure caused by using the system version of `liboctomap-dev` rather than the one provided by the ROS build farm. I suppose there is a patch somewhere between 1.9.3 and 1.9.8 that resolves this issue

marip8, 2023-03-09:
@Levi-Armstrong how do you feel about adding the correct version of `octomap` to the `dependencies.rosinstall`? It's not ideal, but I can't find any later versions distributed on PPAs, etc. It also looks like jammy only provides version 1.9.7, which may also not solve the issue

Levi-Armstrong, 2023-03-09:
I think add in it to the rosinstall is fine. Should we also add it to the CMake find_package for octomap also?

Levi-Armstrong, 2023-03-09:
If this occurs on you machine you might run ldd and see if it is linking against both libraries which can cause weird behaviors like this.

marip8, 2023-03-09:
It looks like it only links against the system version