tesseract-robotics/tesseract #829: Find closest distance of link pairs given a trajectory composite

cpetersmeier, 2022-10-22:
Hello Everyone,

I am trying to evaluate the closest distance of a link that is fixed to the robot to the collision geometries in the environment, given a composite instruction that holds a trajectory. 

I tried the discrete and continuous collision checking task generators but found that they only report contact results if a collision has occurred. Is there another way to do this?

Thanks in advance!

Levi-Armstrong, 2022-10-23:
The lowest level approach would be to get a contact manager from the environment and set the active links and then set the contact distance threshold to something large enough get the data you want. Then perform a contact test and filter the data for the link you are interested in. This can be done using the same utility function for checking the trajectory by doing the same with setting the contact distance threshold to something large and setting it to return all contact data.

cpetersmeier, 2022-10-25:
Thanks for the quick reply! I am not sure which method to use to set the contact distance threshold. I am looking at the DiscreteContactManager, would setting the default collision margin via the ContactManagerConfig be the correct way?

Levi-Armstrong, 2022-10-25:
> Thanks for the quick reply! I am not sure which method to use to set the contact distance threshold. I am looking at the DiscreteContactManager, would setting the default collision margin via the ContactManagerConfig be the correct way?

Yea, that would be the correct way.

cpetersmeier, 2022-11-04:
Thanks a lot for your help and the quick reply, it works great!

closed 2022-11-04 by cpetersmeier