tesseract-robotics/tesseract #830: how create

lanmo0923, 2022-10-27:
thank you

Levi-Armstrong, 2022-10-28:
You might have better luck asking this on the tesseract_python repo. @johnwason has been maintaining the python wrappers there. There may be examples.

johnwason, 2022-10-31:
@lanmo0923 did you figure out the problem? Typically the collision manager is created from the environment.

https://gist.github.com/johnwason/275eb1cd0655f9a92f523ce539b904bf

This example is a little out of date though, the resource loader is not going to work. @Levi-Armstrong are there some generic resource loaders in the core library now?

In this case, the SRDF file contains the links to the plugin file for collision checking. 

https://github.com/tesseract-robotics/tesseract/blob/015a9eb24b4f0eb2f3f6e4ae9f763976c49f4a8a/tesseract_support/urdf/lbr_iiwa_14_r820.srdf#L48

Levi-Armstrong, 2022-11-01:
Yea, I just added one which extracts information from environment variables in version 0.14.0.

Levi-Armstrong, 2022-12-27:
No response

closed 2022-12-27 by Levi-Armstrong